analogdevicesinc/scopy #617: Need to have the Scope stoped before doing auto cal

damercer, 2019-06-05:
Hi:
Thia issue was raised on engineer zone but since it is might be something to change in the software I'm re-posting it here. A user has noticed that the scope (and maybe the signal generator?) needs to be stopped before doing an (on demand) auto calibration. Probably need Scopy to check if everything is stopped or not (stopping things as needed) before doing the auto cal routine.

"I'll add that it appears that you want to stop the scope when doing the automatic calibration."

https://ez.analog.com/university-program/f/q-a/111853/adalm2000-expected-drift-of-function-generator

Doug

adisuciu, 2019-07-12:
Fixed with https://github.com/analogdevicesinc/scopy/pull/621

amiclaus, 2020-01-16:
[image]
The UI shows that the instruments are running while calibration is ongoing.

tagoylo, 2020-10-21:
Able to calibrate while instruments are running. This enhancement is implemented. Closing this item.

closed 2020-10-21 by tagoylo