analogdevicesinc/scopy #622: Time alignment of DAC channels when using oversampling

adisuciu, 2019-06-24:
Setup:
Load a csv file 
Load a waveform
Shift phase of the waveform 

Expected result:
Phase shift is smooth

Phase shift is actually random.

On further investigation, it seems that even though the signals start at the correct phase shifts, the signals are not started synced.

Reference:
CSV File:
```
0.0
0.1
0.2
0.3
0.4
0.5
0.6
0.7
0.8
0.9
```
[image]

adisuciu, 2019-09-25:
Should be fixed with latest firmware

HRosete, 2020-10-21:
Phase shift issue fixed at Scopy v1.2.0 (verified at 33ca72f9), FW:[m2k-fw-v0.25-3-g1c38-dirty.zip](https://github.com/analogdevicesinc/scopy/files/5414289/m2k-fw-v0.25-3-g1c38-dirty.zip). Closing this issue.

closed 2020-10-21 by HRosete