analogdevicesinc/scopy #624: Scopy hangs when connecting on windows 7

pantechsolutions, 2019-06-26:
Hello

i am using windows 7,Scopy hangs while connecting the device.it just display calibrating and then it hangs on.

Environmetal variable setting is corrrect.Scopy can detect adalm2000 .
but when click connect it just started calibrating and then  it hangs.

any help,thanks

adisuciu, 2019-06-26:
Did it work before , or is this just happening ? Does the board work on a different computer ?

Make sure you have these drivers installed: https://github.com/analogdevicesinc/plutosdr-m2k-drivers-win/releases
Make sure your restart your computer after installing Scopy. 
If all else fails, can you try connecting via ip: 192.168.2.1 ?

pantechsolutions, 2019-06-26:
i am reseller ,i sold out more than 50 customer so far,it worked fine.
 when i installed in one of my customer computer which has windows 7,it was not working.

i have download and install latest version.
I have restart.
I have check environmental variables
i have also cleared files in app data

one think i noticed some different was the screenshot
[image]
in universal serial bus devices ,the working computer has an IIO,Whereas in this pc,it was mentioned m2k iio usbd.

pantechsolutions, 2019-06-26:
If all else fails, can you try connecting via ip: 192.168.2.1 ?

This also fails.

adisuciu, 2019-06-26:
Can you download this -  https://github.com/analogdevicesinc/libiio/releases/download/v0.18/libiio-0.18.g4e22517-Windows.zip

Run a command prompt, navigate to C:\...\libiio-0.18.g4e22517\MS64 (or MS32 depending on your OS version), and run iio_info -a and send us the output of that command ?

pantechsolutions, 2019-06-26:
[image]

pantechsolutions, 2019-06-26:
[image]

pantechsolutions, 2019-06-26:
[image]

pantechsolutions, 2019-06-26:
Those are the errors while try to installing IIO in windows 7

pantechsolutions, 2019-06-26:
[image]

pantechsolutions, 2019-06-27:
Hello any help on this issue

pantechsolutions, 2019-06-27:
Hello sir, Am getting different results while running the command which u told. But still its closing automatically while calibrating with the device.
[image]

[image]

adisuciu, 2019-06-27:
Hello,

It looks like the USB connection to the M2k is working. Can you modify in C:\Users\welcome\AppData\Roaming\ADI\scopy.ini -> skip_calibration=true . Try to reconnect, see if it fails. The device should not work correctly(it should not give out the correct voltages) but we can diagnose further if there is a calibration issue or not.

-Adrian

pantechsolutions, 2019-06-27:
There is nothing present inside that scopy.ini file

adisuciu, 2019-06-27:
Try running Scopy, and closing it correctly without connecting to the board. That should populate Scopy.ini . Also replace <welcome> from the path with your username.

-Adrian

pantechsolutions, 2019-06-28:
Hai, in some pc its not present. In some its present but still its for calibration. While calibrating oscilloscope. It Closes

pantechsolutions, 2019-06-29:
Hi, Its urgent, we bought more products, ts essential to fix this issue soon. Actual problem is, in some system it is showing that ADI directory, but in some system Not. Then even if we changed the Skip calibration to True still after connecting ts showing same error

adisuciu, 2019-07-01:
Hello, 

Unfortunately we cannot reproduce your issue. Please try uninstalling the driver and Scopy and then reinstall. You can uninstall it from add/remove programs. Also try updating the board firmware to the latest version.
https://wiki.analog.com/university/tools/pluto/users/firmware

If things work after the reinstall don't forget to remove the file mentioned previously so things start out fresh.

-Adrian

closed 2019-09-25 by adisuciu