analogdevicesinc/scopy #628: Network Analyzer some bugs

ciuffoly, 2019-07-01:
Here 2 different problems found in Scopy function Network Anallyzer.

1) the cursor show the correct frequency but bad dB, there are only integer and no decimal

2) the db axe does not show decimal so many time the same db and sometimes I see 16,17,18,18,19,19,20 in the scale

[image]

[image]

ciuffoly, 2019-07-01:
To reproduce the problem just expand the dB scale to 1dB step or less

ciuffoly, 2019-07-01:
At this this I am using only Windows version so I don't' know if the problem exist on other OS.

adisuciu, 2019-07-01:
You are correct, we missed the decimal point in the implementation. We'll fix this.

ciuffoly, 2019-07-01:
Which is the correct setting for the level and the buffer size ?  I see a correct frequency response only with High level and any buffer size (ms)

adisuciu, 2019-07-15:
Fixed with https://github.com/analogdevicesinc/scopy/pull/642

ciuffoly, 2019-07-16:
Is necessary to download and install a new release ?

adisuciu, 2019-07-16:
It's in testing until we create a new release. Assuming you use windows 64bit, you can download a build artifact from our CI. However it's not an official release. - https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/25988096/job/6f6d4nymm4dlg4uu/artifacts - download scopy-64-setup.exe

ciuffoly, 2019-07-16:
Have you seen the problem of db if you reduce db scale ?

the db axe does not show decimal but I see 16,17,17,18,18,19,19,20 in the scale

adisuciu, 2019-07-16:
Yes, it is fixed.

ciuffoly, 2019-07-16:
the new scopy-64-setup.exe or the zip file give an error at the start on libma...9.dll

ciuffoly, 2019-08-02:
Very good result with the new release

[image]
[image]

andreeaipop, 2019-08-30:
On Scopy build: d198b9a (v1.0.6) the bug is fixed.

closed 2019-08-30 by andreeaipop