analogdevicesinc/scopy #629: Network Analyzer overlap

ciuffoly, 2019-07-01:
Is possible to add a new button to save the current plot and change the color of the next measurement to see the differences like in this image create with Paint ?

[image]

adisuciu, 2019-07-01:
That would be a good enhancement. We have reference waveforms for the oscilloscope instrument. You're suggesting the same for the network analyzer.

DanielGuramulta, 2019-07-24:
Implemented with: #643

ciuffoly, 2019-07-31:
Where I can download the new version ?

DanielGuramulta, 2019-07-31:
You can download the installer for the latest build from: https://wiki.analog.com/university/tools/m2k/scopy

ciuffoly, 2019-07-31:
There is no overlap button on this release

DanielGuramulta, 2019-08-01:
On: https://wiki.analog.com/university/tools/m2k/scopy you will find 3 download links for Scopy, one for the latest release,  and two for the latest nightly builds (one for 32-bit systems and one for 64-bit systems) 
[image]
Select the second or third installer depending on your system.
Once you finish installing Scopy. Open the Network Analyzer and on the right top corner select the "wheel" menu. There you will find options to import from csv or txt files already exported sweeps, there will also be a "Snapshot" button which will add as a reference the sweep that is currently on the plot, and there you will also find a "Remove Reference" button which will clear the screen.
[image]

ciuffoly, 2019-08-02:
Very good result

[image]

ciuffoly, 2019-08-02:
Should be possible add more import than one plot using with different colors ?

rgetz, 2019-08-05:
One of the things we are trying to do - is keep things accessible.

Red-green color blindness effects as many as 8 percent of men and 0.5 percent of women with Northern European ancestry.
https://www.colour-blindness.com/general/prevalence/

the other things we are trying to do, is not re-create GNU Plot, Excel, or MATLAB. There are lots of plotting packages...

What is the use case for multiple plots?

Thanks

ciuffoly, 2019-08-05:
just to see the effect of a variations on an amplifier like the load

andreeaipop, 2019-08-30:
On Scopy build: d198b9a (v1.0.6) the enhancement is fixed.

closed 2019-08-30 by andreeaipop