analogdevicesinc/scopy #630: Network Analyzer with Low and Automatic gain bad result

ciuffoly, 2019-07-01:
Why the Network Analyzer give a good result only with High gain mode ?

Low and Automatic gain give a bad frequency response also using a super simple RC filter with 500ohm 6800pF.

[image]

[image]

adisuciu, 2019-07-01:
We'll reproduce this and look into this

ciuffoly, 2019-07-01:
High gain mode increase the reference voltage or only the other channel ?

adisuciu, 2019-07-01:
It shouldn't increase the reference, just set the gain mode of the input channel accordingly - you won't be able to analyse circuits with responses > 2.5V . Automatic should hop in between the input ranges (25V / 2.5V) based on previous circuit response.

ciuffoly, 2019-08-02:
Very good result now, Low, Automatic and High quite the same plot

[image]

ciuffoly, 2019-08-13:
Still some problem with auto, low and high

[image]

andreeaipop, 2019-08-30:
On Scopy build: d198b9a (v1.0.6) the bug is fixed.

adisuciu, 2019-09-25:
Is this still a problem ? Can you share your test setup ?

ciuffoly, 2019-09-26:
[
[image]
[image]

](url)

ciuffoly, 2019-09-26:
with 1V ref
[image]

mhennerich, 2019-09-26:
Hmm - Scopy version 1.1.0 released and we also have a new firmware - you still seem to be using 1.0.6...

ciuffoly, 2019-09-27:
ok, Scope v1.1.1

phase better but not perfect

still many difference from auto (red) to high (orange)

[image]

ciuffoly, 2019-09-27:
Using 500mV insetad of 1V the is no problems, input saturate ?

[image]

ciuffoly, 2019-09-27:
I am testing a Gainclone LM3886

ciuffoly, 2019-09-27:
Using for the phase the range +180 and -180 and 500mV all is ok in auto and high mode

[image]

ciuffoly, 2019-09-27:
Why with 1V there are problems, the input saturate ?

ciuffoly, 2019-09-27:
[image]

adisuciu, 2019-09-30:
Yes, high gain mode has input range of +/- 2.5 V. 
You are saturating the input when using 1V amplitude for the reference waveform to the amplifier.

ciuffoly, 2019-09-30:
Could you recognize this saturation and avoid displaying the wrong graph ?

adisuciu, 2019-09-30:
We probably can. We'll add a feature for this next release.

ciuffoly, 2019-10-01:
All new info has been published

http://www.audiodesignguide.com/AudioMesurement/index.html

closed 2020-07-16 by adisuciu