analogdevicesinc/scopy #632: Compile problem

ciuffoly, 2019-07-02:
I have followed the steps of page

http://www.msys2.org/   +
https://wiki.analog.com/university/tools/m2k/scopy/build-windows

but I have this error

wget "https://ci.appveyor.com/api/projects/analogdevicesinc/scopy-mingw-build-deps/artifacts/scopy-mingw64-build-deps.tar.xz?branch=master&job=Environment: MINGW_VERSION=mingw64, ARCH=x86_64" -O- | tar -C / --strip-components=1 -xJ
--2019-07-02 15:04:57--  https://ci.appveyor.com/api/projects/analogdevicesinc/scopy-mingw-build-deps/artifacts/scopy-mingw64-build-deps.tar.xz?branch=master&job=Environment:%20MINGW_VERSION=mingw64,%20ARCH=x86_64
tar.exe: Error opening archive: Failed to open '\\.\tape0'
Resolving ci.appveyor.com (ci.appveyor.com)... 138.91.141.243
Connecting to ci.appveyor.com (ci.appveyor.com)|138.91.141.243|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://appveyorcidatav2.blob.core.windows.net/analogdevicesinc-17855/scopy-mingw-build-deps/1-0-244/yeas6b33iicihfcm/scopy-mingw64-build-deps.tar.xz?sv=2015-12-11&sr=c&sig=J4ZOP4QkDVHtUOQOIzzAI5lTdfA0cUQctzxumTP7zuE%3D&st=2019-07-02T13%3A03%3A59Z&se=2019-07-02T13%3A09%3A59Z&sp=r [following]
--2019-07-02 15:04:59--  https://appveyorcidatav2.blob.core.windows.net/analogdevicesinc-17855/scopy-mingw-build-deps/1-0-244/yeas6b33iicihfcm/scopy-mingw64-build-deps.tar.xz?sv=2015-12-11&sr=c&sig=J4ZOP4QkDVHtUOQOIzzAI5lTdfA0cUQctzxumTP7zuE%3D&st=2019-07-02T13%3A03%3A59Z&se=2019-07-02T13%3A09%3A59Z&sp=r
Resolving appveyorcidatav2.blob.core.windows.net (appveyorcidatav2.blob.core.windows.net)... 52.239.162.164
Connecting to appveyorcidatav2.blob.core.windows.net (appveyorcidatav2.blob.core.windows.net)|52.239.162.164|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5342972 (5.1M) [application/octet-stream]
Saving to: ‘STDOUT’

-                                 0%[                                                        ]       0  --.-KB/s    in 0s

Cannot write to ‘-’ (No error).

and after also this error

$ cmake -GNinja ~/scopy
-- The CXX compiler identification is GNU 9.1.0
-- Check for working CXX compiler: D:/msys64/mingw64/bin/c++.exe
-- Check for working CXX compiler: D:/msys64/mingw64/bin/c++.exe -- broken
CMake Error at D:/msys64/mingw64/share/cmake-3.14/Modules/CMakeTestCXXCompiler.cmake:53 (message):
  The C++ compiler

    "D:/msys64/mingw64/bin/c++.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: D:/msys64/home/ciuffoly/scopy/CMakeFiles/CMakeTmp

    Run Build Command(s):D:/msys64/usr/bin/ninja.exe cmTC_30dbe
    [1/2] Building CXX object CMakeFiles/cmTC_30dbe.dir/testCXXCompiler.cxx.obj
    FAILED: CMakeFiles/cmTC_30dbe.dir/testCXXCompiler.cxx.obj
    D:\msys64\mingw64\bin\c++.exe     -o CMakeFiles/cmTC_30dbe.dir/testCXXCompiler.cxx.obj -c testCXXCompiler.cxx
    /bin/sh: D:msys64mingw64binc++.exe: command not found
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:19 (project)

-- Configuring incomplete, errors occurred!
See also "D:/msys64/home/ciuffoly/scopy/CMakeFiles/CMakeOutput.log".
See also "D:/msys64/home/ciuffoly/scopy/CMakeFiles/CMakeError.log".

AlexandraTrifan, 2019-07-03:
Hi,

Can you try to run **pacman -Syu** again to check if there are any remaining updates or any issues?
For the first issue, with the download and archive, please try to reinstall tar, using: **pacman -S tar** and check the command again. 

If it still does not work, you can download this  https://ci.appveyor.com/api/projects/analogdevicesinc/scopy-mingw-build-deps/artifacts/scopy-mingw64-build-deps.tar.xz?branch=master&job=Environment:%20MINGW_VERSION=mingw64,%20ARCH=x86_64 . In the downloaded archive, you will find a folder named **"msys64"** . Just copy this over your **D:/msys64** . If a Windows pop up appears, just merge the two folders. In the downloaded archive you have some required dependencies for Scopy, that will be copied in the required location.

For the second part of the issue:
Please try to run **gcc --version** or **g++ --version** , and maybe try to reinstall gcc using **pacman -S mingw-w64-x86_64-gcc**

-Alexandra

ciuffoly, 2019-07-03:
I have reinstalled all from zero with the same procedure  and now I am near the end but still a little problem.

BREAKPADCLIENT_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_INCLUDE_DIRS
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy

-- Configuring incomplete, errors occurred!
See also "D:/msys64/home/ciuffoly/scopy/build/CMakeFiles/CMakeOutput.log".

ciuffoly, 2019-07-03:
sorry this output

GNURADIO_IIO_FOUND = TRUE
-- Boost version: 1.70.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   chrono
--   date_time
--   atomic
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Doxygen not found, API documentation won't be generated
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
BREAKPADCLIENT_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_INCLUDE_DIRS
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy

-- Configuring incomplete, errors occurred!
See also "D:/msys64/home/ciuffoly/CMakeFiles/CMakeOutput.log".

ciuffoly, 2019-07-03:
after    pacman -S doxygen

I still have this error

-- Found Doxygen: D:/msys64/usr/bin/doxygen.exe (found version "1.8.15") found components:  doxygen missing components:  dot
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
BREAKPADCLIENT_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_INCLUDE_DIRS
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy

-- Configuring incomplete, errors occurred!
See also "D:/msys64/home/ciuffoly/scopy/build/CMakeFiles/CMakeOutput.log".

ciuffoly@DESKTOP-CH7EA9H MINGW64 ~/scopy/build

dNechita, 2019-07-03:
Try:
pacman -S graphviz

ciuffoly, 2019-07-03:
$ pacman -S graphviz
error: target not found: graphviz

dNechita, 2019-07-03:
pacman -S mingw-w64-graphviz

ciuffoly, 2019-07-03:
$ pacman -S mingw-w64-graphviz
error: target not found: mingw-w64-graphviz

ciuffoly, 2019-07-03:
start this ?

pacman -S mingw-w64-i686-graphviz

dNechita, 2019-07-03:
or mingw-w64-x86_64-graphviz

ciuffoly, 2019-07-03:
same error

GNURADIO_IIO_FOUND = TRUE
-- Boost version: 1.70.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   chrono
--   date_time
--   atomic
-- Found Doxygen: D:/msys64/usr/bin/doxygen.exe (found version "1.8.15") found components:  doxygen dot
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
BREAKPADCLIENT_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_INCLUDE_DIRS
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy

-- Configuring incomplete, errors occurred!
See also "D:/msys64/home/ciuffoly/scopy/CMakeFiles/CMakeOutput.log".

AlexandraTrifan, 2019-07-03:
Please run again the command in the following way: 
**cmake -GNinja -DBREAKPAD_HANDLER=OFF ~/scopy**

The bug here is that it should probably be OFF by default. We will also add it to the documentation as soon as possible.

-Alexandra

ciuffoly, 2019-07-03:
$ cmake -GNinja -DBREAKPAD_HANDLER=OFF ~/scopy
-- QWT Version: 6.4.0
-- QWT Version must be >= 5.2 and <= 6.2.0, Found 6.4.0
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: ANALOG
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-analog.dll.a;D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a;D:/msys64/mingw64/lib/libvolk.dll.a
GNURADIO_ANALOG_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-blocks.dll.a;D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: FFT
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-fft.dll.a;D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_FFT_FOUND = TRUE
Checking for GNU Radio Module: FILTER
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-filter.dll.a;D:/msys64/mingw64/lib/libgnuradio-fft.dll.a;D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: PMT
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_PMT_FOUND = TRUE
Checking for GNU Radio Module: VOLK
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libvolk.dll.a
GNURADIO_VOLK_FOUND = TRUE
Checking for GNU Radio Module: IIO
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-iio.dll.a;D:/msys64/mingw64/lib/libgnuradio-analog.dll.a;D:/msys64/mingw64/lib/libgnuradio-blocks.dll.a;D:/msys64/mingw64/lib/libgnuradio-analog.dll.a;D:/msys64/mingw64/lib/libvolk.dll.a;D:/msys64/mingw64/lib/libgnuradio-blocks.dll.a;D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_IIO_FOUND = TRUE
-- Boost version: 1.70.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   chrono
--   date_time
--   atomic
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
BREAKPADCLIENT_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_INCLUDE_DIRS
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy

-- Configuring incomplete, errors occurred!
See also "D:/msys64/home/ciuffoly/scopy/CMakeFiles/CMakeOutput.log".

ciuffoly@DESKTOP-CH7EA9H MINGW64 ~/scopy

ciuffoly, 2019-07-03:
Do you want a teamview session ?

AlexandraTrifan, 2019-07-03:
Please clean up everything in the **build** folder and try the command again .

ciuffoly, 2019-07-03:
ciuffoly@DESKTOP-CH7EA9H MINGW64 ~/scopy
$ rm -rf build/

ciuffoly@DESKTOP-CH7EA9H MINGW64 ~/scopy
$ mkdir ~/scopy/build

ciuffoly@DESKTOP-CH7EA9H MINGW64 ~/scopy
$ cmake -GNinja -DBREAKPAD_HANDLER=OFF ~/scopy
-- QWT Version: 6.4.0
-- QWT Version must be >= 5.2 and <= 6.2.0, Found 6.4.0
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: ANALOG
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-analog.dll.a;D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a;D:/msys64/mingw64/lib/libvolk.dll.a
GNURADIO_ANALOG_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-blocks.dll.a;D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: FFT
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-fft.dll.a;D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_FFT_FOUND = TRUE
Checking for GNU Radio Module: FILTER
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-filter.dll.a;D:/msys64/mingw64/lib/libgnuradio-fft.dll.a;D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: PMT
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_PMT_FOUND = TRUE
Checking for GNU Radio Module: VOLK
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libvolk.dll.a
GNURADIO_VOLK_FOUND = TRUE
Checking for GNU Radio Module: IIO
 * INCLUDES=D:/msys64/mingw64/include
 * LIBS=D:/msys64/mingw64/lib/libgnuradio-iio.dll.a;D:/msys64/mingw64/lib/libgnuradio-analog.dll.a;D:/msys64/mingw64/lib/libgnuradio-blocks.dll.a;D:/msys64/mingw64/lib/libgnuradio-analog.dll.a;D:/msys64/mingw64/lib/libvolk.dll.a;D:/msys64/mingw64/lib/libgnuradio-blocks.dll.a;D:/msys64/mingw64/lib/libgnuradio-runtime.dll.a;D:/msys64/mingw64/lib/libgnuradio-pmt.dll.a
GNURADIO_IIO_FOUND = TRUE
-- Boost version: 1.70.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   chrono
--   date_time
--   atomic
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
BREAKPADCLIENT_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_INCLUDE_DIRS
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
   used as include directory in directory D:/msys64/home/ciuffoly/scopy
BREAKPAD_LIBRARIES
    linked by target "scopy" in directory D:/msys64/home/ciuffoly/scopy

-- Configuring incomplete, errors occurred!
See also "D:/msys64/home/ciuffoly/scopy/CMakeFiles/CMakeOutput.log".

ciuffoly@DESKTOP-CH7EA9H MINGW64 ~/scopy

ciuffoly, 2019-07-03:
solved the problem with:
rm -rf scopy
mkdir ~/scopy/build
cd ~/scopy/build
cmake -GNinja -DBREAKPAD_HANDLER=OFF ~/scopy

now I have

-- Configuring done
-- Generating done
-- Build files have been written to: D:/msys64/home/ciuffoly/scopy/build

ciuffoly, 2019-07-03:
Why if I change the source src/tool_launcher.cpp the make does not recompile this ?

AlexandraTrifan, 2019-07-04:
Great. Let us know if you encounter any other issues.

closed 2019-07-04 by AlexandraTrifan