analogdevicesinc/scopy #635: GUI woes with scopy on macos

phjo33, 2019-07-06:
OS Version: macOS Mojave (10.14)
FW Version: v0.22
SW Version: 48fb6a9

Description of the bug:Scopy is unusable on macos

Running on a retina macbook from 2012, with a rather fresh install of mac os mojave 10.14.5, scopy is about unusable.

Numerous little graphic glitches with strings and symbols partially hidden, some very minor (trigger choices for the logic analyser, where only a small part of the symbol is shown), some more annoying (measured values where the first digit is not showing, for example with cursors)

I could find any way to navigate through the timeline for the Logic Analyzer : no key seems to have any action, no visible slider, and the touchpad on the macbook is much too sensible to be of any use. If I use two fingers vertical movement (simulating middle scroll) then in a instant I completely lose focus of where I was, going from a 1s division to divisions the order of picoseconds ! Behaviour almost seem random.

As soon as one or more tools are in use, GUI very often becomes quite unresponsive. Usually, I just have to force-quit scopy as it seems impossible to even stop the scope, the analyzer or do just anything.

Having tried scopy on my old T430 thinkpad runing linux mint, behaviour is much better there. Gui stays responsive, scopy seems just much faster in every regard (whereas the T430 is not as powerful as the macbook is). So the problem is not the M2K hardware, but the software.

AlexandraTrifan, 2020-09-17:
Tested this on macOS Catalina 10.15 using the following build ( https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/35251245/job/oxx7gqvvslv7gf19/artifacts ). This is the prepared release candidate build.

This version includes a revamped Logic Analzyer: the zooming is now done using a mouse defined square (as it was done in the Oscilloscope; allows more precision), cursor readouts are no longer overlapped with any other UI elements. Overall the UI experience on macOS was improved with all the latest updates.

adisuciu, 2020-10-21:
Closing this, reopen if you are still having issues.

closed 2020-10-21 by adisuciu