analogdevicesinc/scopy #651: libmatio2 support

jpnielsen, 2019-07-15:
I'm following https://github.com/analogdevicesinc/scopy/wiki/Building-Scopy-on-Linux

It seems that libmatio2 is not available in later ubuntu or debian stable releases.

https://packages.ubuntu.com/search?keywords=libmatio2
https://packages.debian.org/search?suite=all&arch=any&searchon=names&keywords=libmatio2

adisuciu, 2019-07-16:
Can you try installing libmatio-dev ? or libmatio4. We'll update with whatever works.

jpnielsen, 2019-07-16:
neither are in ubuntu/eoan64 - they were in ubuntu/disco64

https://packages.ubuntu.com/search?keywords=libmatio&searchon=names&suite=disco&section=all

jpnielsen, 2019-07-16:
I tried compiling using libmatio4, but then again...

In the install guide you ask us to install old versions of libraries .. I'm not going to do that just to compile scopy - too much hassle - I'm going to use the nbinary releases instead.

libboost is version 1.70 now
libvolk is 1.4 now 
libsigrok is 0.5.1 now 

etc.

I look forward to a serious update of scopy, using current libraries.

jpnielsen, 2019-07-16:
Followed the webpage, using 
`sudo apt-get install git cmake libzip-dev autoconf libtool libxml2 libxml2-dev libmatio4 libmatio-dev`

i got to:

`:~/scopy/build$ cmake ..`

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
BREAKPADCLIENT_LIBRARIES
    linked by target "scopy" in directory /home/vagrant/scopy
BREAKPAD_INCLUDE_DIRS
   used as include directory in directory /home/vagrant/scopy
   used as include directory in directory /home/vagrant/scopy
   used as include directory in directory /home/vagrant/scopy
   used as include directory in directory /home/vagrant/scopy
   used as include directory in directory /home/vagrant/scopy
   used as include directory in directory /home/vagrant/scopy
   used as include directory in directory /home/vagrant/scopy
BREAKPAD_LIBRARIES
    linked by target "scopy" in directory /home/vagrant/scopy

-- Configuring incomplete, errors occurred!
```
but breakpad doesn't compile on gcc-9 out-of-the box.

```
cd ~ && git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git
cd depot_tools/
export PATH=$PATH:~/depot_tools
mkdir ~/breakpad
cd ~/breakpad
fetch breakpad
cd src
./configure
make
```
I'm new to compilers, and I don't know how to fix this error:

`In file included from ./src/common/memory_allocator.h:50,
                 from ./src/client/linux/dump_writer_common/thread_info.h:37,
                 from ./src/client/linux/minidump_writer/linux_dumper.h:54,
                 from ./src/client/linux/minidump_writer/minidump_writer.h:41,
                 from ./src/client/linux/handler/exception_handler.h:42,
                 from src/client/linux/handler/exception_handler.cc:66:
./src/third_party/lss/linux_syscall_support.h: In member function ‘bool google_breakpad::ExceptionHandler::GenerateDump(google_breakpad::ExceptionHandler::CrashContext*)’:
./src/third_party/lss/linux_syscall_support.h:2408:75: error: listing the stack pointer register ‘rsp’ in a clobber list is deprecated [-Werror=deprecated]
 2408 |                              : "rsp", "memory", "r8", "r10", "r11", "rcx");
      |                                                                           ^
./src/third_party/lss/linux_syscall_support.h:2408:75: note: the value of the stack pointer after an ‘asm’ statement must be the same as it was before the statement
cc1plus: all warnings being treated as errors
make: *** [Makefile:5154: src/client/linux/handler/exception_handler.o] Error 1
`

jpnielsen, 2019-07-16:
Managed to compile google_breakpad

make CXXFLAGS='-Wno-error=deprecated'
sudo make install

cd ~scopy && make

jpnielsen, 2019-07-16:
In debian 10, before building qwt

```
 sudo apt install qt5-qmake
 sudo apt install qt5-default 
```

jpnielsen, 2019-07-17:
I have a scopy build on debian 10.0.0 now

```
adduser mylogin dialout
sudo ldconfig

sudo apt install screen
screen /dev/ttyACM0
```
gives me a m2k prompt.

I can't see the m2k device in scopy.

rgetz, 2019-07-17:
thanks for the report - I think @adisuciu is on vacation for a few days.

Normally the udev rules are installed either in the deb package, or in the "sudo make install".

Need to be careful that the udev rule can find iio_info in it's path. (If you install iio_utils in a non-standard place, it will silently fail, and you can can run as root, but not as a normal user),

rgetz, 2019-07-17:
& the reason we don't use the Mathworks libraries that you point to - is they are only installed on end users systems when those end users have MATLAB installed; and are not license compatible with the rest of scopy. (like matio is - which is under a Simplified BSD License).

closed 2020-04-24 by jpnielsen