analogdevicesinc/scopy #652: <Instrument>: <Short description of the bug>

RonFredericks, 2019-07-17:
OS Version: Windows 10 (10.0)
FW Version: 
SW Version: 48fb6a9

Description of the bug:<Text within scopy program are way too small, and barely readable. This issue appears when using Microsoft's Display Setting "scaling factor" settings. The "scaling factor for my PC is set to 350% so I can use the native monitor settings for PC of 3840x2160. Without the scaling factor I would have to use a lower screen resolution with text now blurry, or all text and icons would be too small.

So please include support for windows  Display Settings -> Scale Factor; as most modern programs now support.>

Steps to reproduce:
- Right mouse click on blank area of monitor and select Display Settings.
- Select a scaling factor of 350% as an example
- Again right click mouse on blank area of monitor and select tool for setting monitor resolution, in my case it is NVIDIA Control Panel. 
 - Select highest resolution, in my case indicated as (native) 3840 x 2160.
- Launch Scopy to see that text is almost not readable because it is too small.

My hardware:
 - NVIDIA Titan V graphics card (but any gaming card should work as well for this bug demonstration)
 - Samsung U28H750 4K QLED   28" 3840x2160 monitor

Some images to demonstrate the issue...

[image]
[image]
[image]
[image]
[image]

--

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/Ron/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

AlexandraTrifan, 2019-07-18:
Hello,

Thank you for your report. We are working on some fixes/enhancements related to this. Will probably include them in the next release. ( https://github.com/analogdevicesinc/scopy/issues/551 )

Thank you

-Alexandra

DanielGuramulta, 2019-07-31:
With pr: #663 there should no longer be an issue. Scopy should look nice on 4K.

amiclaus, 2019-08-30:
On Scopy build: d198b9a (v1.0.6) the bug is fixed.

closed 2019-08-30 by amiclaus