analogdevicesinc/scopy #654: windows nightly version 1.0.2851 libmatio-9.dll error

jpnielsen, 2019-07-22:
I uninstalled scopy, and manually deleted the left over "decoders" dir.
I then installed Windows Nightly - version 1.0.2851 - and rebooted.

Scopy.exe comes up with the error:

"The procedure entry point H5Rdereference could not be located in the dynamic link library D:\scopy\libmatio-9.dll."

jpnielsen, 2019-07-22:
Followed the "build scopy on windows" here: 
https://github.com/analogdevicesinc/scopy/wiki/Building-Scopy-on-Windows
added `pacman -S mingw-w64-x86_64-qt5 --disable-download-timeout` found on ez.analog.com

This is the error i get when trying to run "scopy"

> "Scopy.exe - Entry Point Not Found"
> "The procedure entry point H5Rdereference could not be located in the dynamic link library D:\msys64\mingw64\bin\libmatio-9.dll."

jpnielsen, 2019-07-22:
https://packages.msys2.org/package/mingw-w64-x86_64-matio?repo=mingw64

AlexandraTrifan, 2019-07-23:
Hi, 

Thank you for this. 
We will look into it and fix the build as soon as possible.

-Alexandra

jpnielsen, 2019-07-26:
windows build needs

pacman -S mingw-w64-x86_64-breakpad-git

DanielGuramulta, 2019-07-30:
Windows nightly builds should be fixed with: #660

jpnielsen, 2019-07-30:
Please also update the doc on 

https://github.com/analogdevicesinc/scopy/wiki/Building-Scopy-on-Windows
and
https://wiki.analog.com/university/tools/m2k/scopy/build-windows

DanielGuramulta, 2019-07-31:
I've updated the documentation on both github and wiki.

jpnielsen, 2019-08-01:
I can confirm that scopy builds.
thank you.

minor detail:

from the documentation:

> Finally, add the “SCOPY_PYTHONPATH” variable:
> 
> ```
> Click on “New…”
> In “Variable name:”, write SCOPY_PYTHONPATH
> In “Variable value:” write C:\msys64\mingw64\lib\python3.7;C:\msys64\mingw64\lib\python3.7\plat-win;C:\msys64\mingw64\lib\python3.7\lib-dynload;C:\msys64\mingw64\lib\python3.7\site-packages
> Then click OK.
> ```
> 

"C:\msys64\mingw64\lib\python3.7\plat-win" does not exist on my fresh install of http://repo.msys2.org/distrib/x86_64/msys2-x86_64-20190524.exe

jpnielsen, 2019-08-17:
I don't like the "fix" - it's a likely to fail workaround, and you didn't give an instruction on how to lock the version of hdf5 - first time the user runs an update of msys2, the "fix" is broken.

other software offcause may be dependent on a newer version of hdf5

DanielGuramulta, 2020-01-16:
Since: https://github.com/analogdevicesinc/scopy/commit/42f0b66b753c00c8028696330d30f18f25589de1, we no longer use an older version of hdf5.

closed 2020-01-16 by DanielGuramulta