analogdevicesinc/scopy #656: Oscilloscope External Trigger

christophjurczyk, 2019-07-22:
Hi,

I am currently trying to use the external trigger in Scopy. I connected to TI (1st row 7th column) a 5V square wave (50%@1 Hz) and would like to trigger on the rising edge. 
In Scopy I set the trigger to normal, turned off the analog trigger and set the external trigger to rising edge. Unfortunately it stays in waiting mode forever.

[Scopy](https://i.ibb.co/0hBycX2/Screenshot-from-2019-07-22-15-07-25.png)

Is there something wrong in the configuration or is it limitation?

Thanks for your help!
-Christoph

adisuciu, 2019-07-23:
Hey Christoph,
You should select channel 1 under source if you trigger on TI pin. 

-Adrian

christophjurczyk, 2019-07-23:
Hi Adrian,
I already tried both channels. Same behaviour on both channels in any software/hardware configuration.

[image]

I even swapped the ADALM 2000, but same issue.

adisuciu, 2019-07-24:
Hey, we tracked the issue down. It seems to be an HDL issue where the external trigger is only correctly activated at 100MHz sample rates, but might get decimated on lower. Can you confirm that on lower timebases the Oscilloscope triggers succesfully ?

-Adrian

christophjurczyk, 2019-07-25:
Hey Adrian,
yes I can confirm this. External trigger is only working when 100MHz sample rates are active. 
Good to hear that you can reproduce this issue.
-Christoph

closed 2019-10-08 by AndreiGrozav