analogdevicesinc/scopy #662: Build failure in src/network_analyzer.cpp due to missing set_len method of gr::fft::goertzel_fc Class

balbirthomas, 2019-07-27:
Build of scopy fails despite using a patched version of gnuradio (which phase and trapezoid patch), libqwt (multiaxes git branch), qwtpolar and other dependencies. The build failure occurs when compiling `src/network_analyzer.cpp` ( [lines 1091 and 1092](https://github.com/analogdevicesinc/scopy/blob/8c4c82b7ee0d00ac74d1777c6179963a9702d94b/src/network_analyzer.cpp#L1091) ) and is caused by a missing method (`set_len()`) in the `gr::fft:goertzel_fc` class of `gnuradio-fft` package. No such method is mentioned in the [official documentation](https://www.gnuradio.org/doc/doxygen/classgr_1_1fft_1_1goertzel__fc.html) of `gnuradio-fft` . There does not seem to be such a method in the [Analog Devices fork of gnuradio](https://github.com/analogdevicesinc/gnuradio/blob/master/gr-fft/lib/goertzel_fc_impl.h) either. 

If this is not a bug could you kindly clarify the issue and close this ticket ?

DanielGuramulta, 2019-07-29:
Hi,
The patch for the goertzel_fc block is available on the "scopy" branch of the gnuradio fork: https://github.com/analogdevicesinc/gnuradio/tree/scopy.

balbirthomas, 2019-07-29:
Thank you. That explains it.

closed 2019-07-29 by balbirthomas