analogdevicesinc/scopy #669: <error has no tips>: <Win10 Visual C++ runtime error,but had no any notice ...>

zfysis, 2019-08-12:
OS Version: Windows 10 (10.0)
FW Version: 
SW Version: 6633459

Description of the bug:<description>
Steps to reproduce:
-M2K firmware had updated with anoter win7 computer successfully,but not in Win10;
-Win10 driver version V0.7
-Scopy  version V1.0.6
-plug in m2k,scopy had already recognise the m2k,but click "connect"button,"Network Analyzer" & "Signal Generator"are not drive successfully,and pop a window about "Runtime Error ! Program:C:\Analog Devices\Scopy\Scopy.exe"but had no a message to tell how to get it ok ...

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/Hider/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

AlexandraTrifan, 2019-08-13:
Hi,

After you click "Connect", the Network Analyzer and Signal Generator buttons are hanging with the text "Calibrating..." ?

Could you tell us the firmware version ? You should see it on front page of Scopy, when you select the device.

Also, please attach here the Scopy.ini configuration file : **C:/Users/Hider/AppData/Roaming/ADI/Scopy.ini**  and the files in **C:/Users/Hider/AppData/Roaming/ADI/crashdmp**

Thank you

-Alexandra

zfysis, 2019-08-19:
Hi,thank you very much for attention to this question,files uploaded already.
[m2k error 2019-8-19.pdf](https://github.com/analogdevicesinc/scopy/files/3514531/m2k.error.2019-8-19.pdf)

[ADI.zip](https://github.com/analogdevicesinc/scopy/files/3514538/ADI.zip)
Scopy version : V1.0.6
m2k Firmware Version : V0.23
IIO Version: 0.18
Win 10 Drivers : V0.7

adisuciu, 2019-08-22:
It looks like the archive does not contain the minidump. Can you try the following:

- Reproduce the error
- Don't run Scopy 
- Navigate to  C:/Users/Hider/AppData/Roaming/ADI/crashdmp
- There should be a file with a unique name - something like 9fd98ec-d55c-29e1-4ae067b0-4aaec0d6.dmp
- Pack that file and send it over

zfysis, 2019-08-22:
1. It ·s very eazy to reproduce the error;
2. if clicked "Connet" ,Scopy shows "Calibrating...","Logic Analyzer,Pattern Generator,Digital IO,Power Supply"shows calibra ok,others are shows "Calibrating..."and Scopy hangs ...
3.There are no other files on   C:/Users/Hider/AppData/Roaming/ADI/crashdmp has unique name like ....dmp

zfysis, 2019-09-08:
Hello,
    I reinstall Win10 to solve this problem,Scopy works ok Now,thank you!

closed 2019-09-08 by zfysis