analogdevicesinc/scopy #676: Value becomes 0 when attempting to change unit in phase

tagoylo, 2019-08-19:
Scopy version: 4fe815a0
firmware verson: v0.23

Steps to reproduce:

Set frequency to 500 Hz, sine wave
Set phase to 0 deg
Then change the unit to ms, then add value 1.5 ms, then enter
Then change unit again to degree, when attempting to change the unit to degree the value will become 0. See screen  recording
[image]

DanielGuramulta, 2019-09-02:
Should be fixed with: #692

amiclaus, 2019-09-02:
On Scopy build: 33b6c1c (v1.1.0) the bug still persists.
OS: Windows 10, 64-bit

DanielGuramulta, 2019-09-04:
I fix an error made in: #692 with #695. From what I tested it looks fixed to me. Can you look again into the issue and confirm that the problem is solved this time?

Thank you!

amiclaus, 2019-09-04:
On Scopy build: 214c48 (v1.10) the bug is fixed.

closed 2019-09-04 by amiclaus