analogdevicesinc/scopy #678: Averaging does not work when in Sample type

tagoylo, 2019-08-20:
Scopy version: 4fe815a0
firmware version: v0.23

Steps to reproduce:

Open Spectrum analyzer, set Type to Sample
Run Spectrum analyzer then increase averaging value.
When increasing the average value, there is no change to the plot.

Averaging: **1**
[image]

Averaging:**90**
[image]

dNechita, 2019-08-20:
When setting the Type to "Sample" in the spectrum analyzer, it means that no averaging will be done. It will just show the samples. If no averaging is performed then the value of "Averaging" will not be used. So the behavior is expected.
Indeed, it can be confusing to allow users to change the averaging value like this. Maybe disable this UI control when in "sample" mode?

DanielGuramulta, 2019-09-02:
Should be fixed with: #682

andreeaipop, 2019-09-02:
On Scopy build: d198b9a (v1.0.6) the bug is fixed.

closed 2019-09-02 by andreeaipop