analogdevicesinc/scopy #688: Network Analyzer: When the cursor is placed on the lower right, details about the magnitude difference is cut

HRosete, 2019-08-22:
OS Version: Windows 10 (10.0)
FW Version: v0.23
SW Version: 4fe815a

Bug Description: When the cursor is placed on the lower right, details about the magnitude difference is cut. This only happens in Network Analyzer.

Steps to reproduce:

1. Open Network Analyzer and enable the cursor.
2. Open the cursor settings and place the cursor on the lower right of the plot.
3. Monitor the magnitude details on your default Scopy window and as you maximize it.

[NA_Cursor.zip](https://github.com/analogdevicesinc/scopy/files/3529585/NA_Cursor.zip)

HRosete, 2019-08-22:
Attached are the pictures of the said bug. Thank you.

DanielGuramulta, 2019-08-22:
I couldn't reproduce the issue. Can you please tell me the resolution you have on the display, and also if you are using scaling?

HRosete, 2019-08-23:
My laptop's monitor resolution is 1920x1080 and I used a 125% scaling.
I tried switching the scale to 100% and the magnitude details fully appeared. Switched to other scales and I could not replicate the issue. My apologies.
Closing this item.

closed 2019-08-23 by HRosete