analogdevicesinc/scopy #689: Cannot manually enter a marker's freq position lower than the starting freq of sweep

HRosete, 2019-08-22:
OS Version: Windows 10 (10.0)
FW Version: v0.23
SW Version: 4fe815a

Bug Description: Even when the option "Only search marker peaks in visible domain" is disabled in the Preferences, one cannot manually enter a marker's frequency position lower to the starting frequency of the sweep.

Profile:
[spectrum_analyzer_profile.zip](https://github.com/analogdevicesinc/scopy/files/3529676/spectrum_analyzer_profile.zip)
10Vpp, 500kHz Sine wave

Steps to reproduce:

1. Load attached profile: "spectrum_analyzer.ini."
2. Go to Preferences, **_disable_** the option "Only search marker peaks in visible domain."
3. Connect AWG1 to Scope Ch1+.
4. Turn on Signal Generator. In Spectrum Analyzer, open Markers setting and turn the marker table on. Run Spectrum Analyzer.
5. Click Marker 1 and manually put the Frequency Position to 500kHz (the frequency position automatically goes to 700kHz, the starting frequency of the sweep.)
6. Click Peak (this automatically places the marker on 500kHz).

jpineda3, 2021-04-11:
Still exists in v.026
[spectrum_analyzer_profile.zip](https://github.com/analogdevicesinc/scopy/files/6291291/spectrum_analyzer_profile.zip)

jpineda3, 2021-04-28:
Still exists in v 1.3-rc2.

HRosete, 2022-05-02:
**Environment:**
 - OS Version: Windows 10 (10.0)
 - FW Version: [v0.28](https://github.com/analogdevicesinc/m2k-fw/releases/tag/v0.28)
 - SW Version: cca49b2
 
This issue is still apparent in [Scopy v1.4.0-rc2](https://github.com/analogdevicesinc/scopy/releases/tag/v1.4.0-rc2). Please see the video below:

https://user-images.githubusercontent.com/43806793/166177671-d7138d0e-2124-4516-87c3-295cb16bc8e1.mp4

IonutMuthi, 2022-05-05:
can't reproduce

closed 2022-05-05 by IonutMuthi