analogdevicesinc/scopy #698: Digital tools running buttons remain visible after disconnecting the board

andreeaipop, 2019-09-05:
OS Version: Windows 10 (10.0)
FW Version: v.023
SW Version: 652c691

Description of the bug:
If the board is disconnected while Logic Analizer/ Pattern generator/ Digital I/O are running, the running buttons remain visible.
Steps to reproduce:
Open Scopy, connect m2k and run multiple tools, including Logic Analizer, Pattern generator and Digital I/O
[image]

In Home tab click Disconnect.
The board is disconnected and the tools are no longer active but running icon is still visible
[image]

DanielGuramulta, 2019-09-09:
Issue should be fixed with: #701

andreeaipop, 2019-09-09:
On Scopy build: 3975846 (v1.1.0) the bug is fixed.

closed 2019-09-09 by andreeaipop