analogdevicesinc/scopy #710: Scopy 1.1.0 cannot load libicui18n.58.dylib on startup

todorspetkov, 2019-09-16:
- OS: Mac Mojave 10.14.6 
 - SW 1.1.0

Scopy crashes on startup. Mac error log shows libicui18n.58.dylib cannot be loaded. 

Full mac error log:

Process:               Scopy [47848]
Path:                  /Applications/Scopy.app/Contents/MacOS/Scopy
Identifier:            org.adi.Scopy
Version:               ??? (1.1.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Scopy [47848]

Date/Time:             2019-09-16 19:34:26.802 +0200
OS Version:            Mac OS X 10.14.6 (18G95)

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/libicui18n.58.dylib
  Referenced from: /Applications/Scopy.app/Contents/Frameworks/iio.framework/Versions/0.18/iio
  Reason: image not found

Binary Images:
       0x10b99c000 -        0x10be07fff +org.adi.Scopy (??? - 1.1.0) <209A7E6D-4C92-3C5A-991D-E422AF77BF66> /Applications/Scopy.app/Contents/MacOS/Scopy
       0x10beac000 -        0x10c2e7ff7 +org.qt-project.QtWidgets (5.13 - 5.13.0) <ECCBD5F5-DF37-3697-AD48-9C93AFC3D94D> /Applications/Scopy.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets
       0x10c450000 -        0x10c8a6fff +org.qt-project.QtGui (5.13 - 5.13.0) <1421E0D5-4E95-3F77-A1C1-DF564D401740> /Applications/Scopy.app/Contents/Frameworks/QtGui.framework/Versions/5/QtGui
       0x10c9ad000 -        0x10cec5fff +org.qt-project.QtCore (5.13 - 5.13.0) <4A13C0CC-CAD3-3275-8AE3-0FF22CF005C4> /Applications/Scopy.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore
       0x10cf84000 -        0x10cf88ffb +org.qt-project.QtConcurrent (5.13 - 5.13.0) <76CEBE4E-437E-3865-BB7B-E769B38AB82E> /Applications/Scopy.app/Contents/Frameworks/QtConcurrent.framework/Versions/5/QtConcurrent
       0x10cf8e000 -        0x10d30bfff +org.qt-project.QtQml (5.13 - 5.13.0) <7ED9F8F1-EB62-3BB5-BBCC-BEC274D2B8EE> /Applications/Scopy.app/Contents/Frameworks/QtQml.framework/Versions/5/QtQml
       0x10d3bd000 -        0x10d456ffb +libgnuradio-runtime.3.7.12.0.dylib (0) <B7E79E78-700C-3061-A80D-D445471AA8DA> /Applications/Scopy.app/Contents/Frameworks/libgnuradio-runtime.3.7.12.0.dylib
       0x10d4b8000 -        0x10d4f8ff3 +libgnuradio-pmt.3.7.12.0.dylib (0) <F388D025-8561-3ADC-9740-442263FD53ED> /Applications/Scopy.app/Contents/Frameworks/libgnuradio-pmt.3.7.12.0.dylib
       0x10d527000 -        0x10d567fff +libgnuradio-analog.3.7.12.0.dylib (0) <27B87B92-4C84-32F5-A7B6-766BEB9C9249> /Applications/Scopy.app/Contents/Frameworks/libgnuradio-analog.3.7.12.0.dylib
       0x10d5ba000 -        0x10d715fef +libvolk.1.3.dylib (0) <9E746D4F-CAE2-3D8F-951A-A9C4E06FEC2E> /Applications/Scopy.app/Contents/Frameworks/libvolk.1.3.dylib
       0x10d7c3000 -        0x10d906ffb +libgnuradio-blocks.3.7.12.0.dylib (0) <59902974-135B-3A56-98FD-DA6F053865B2> /Applications/Scopy.app/Contents/Frameworks/libgnuradio-blocks.3.7.12.0.dylib
       0x10da67000 -        0x10da82fff +libgnuradio-fft.3.7.12.0.dylib (0) <C5BDD6A2-A4D2-3815-BE0A-1EDF4F2C0515> /Applications/Scopy.app/Contents/Frameworks/libgnuradio-fft.3.7.12.0.dylib
       0x10da9b000 -        0x10db0ffff +libgnuradio-filter.3.7.12.0.dylib (0) <F980E668-0499-3371-BD3E-BBCA3B93AACE> /Applications/Scopy.app/Contents/Frameworks/libgnuradio-filter.3.7.12.0.dylib
       0x10db6e000 -        0x10dbacfff +libgnuradio-iio.0.dylib (0) <A86BFC23-F6D4-36C0-A201-1AF2E6E785A4> /Applications/Scopy.app/Contents/Frameworks/libgnuradio-iio.0.dylib
       0x10dbe0000 -        0x10dbe0fff +libboost_system-mt.dylib (0) <6B19EE84-CC90-36DC-A5CA-A51D2003BF61> /Applications/Scopy.app/Contents/Frameworks/libboost_system-mt.dylib
       0x10dbe8000 -        0x10dbf7ff7 +libboost_filesystem-mt.dylib (0) <8B0771E3-90D0-32A2-8729-E3A349F39485> /Applications/Scopy.app/Contents/Frameworks/libboost_filesystem-mt.dylib
       0x10dc0e000 -        0x10dc1bfff +libboost_thread-mt.dylib (0) <5814BFE7-FB6C-3690-A6D1-A1271EC64A26> /Applications/Scopy.app/Contents/Frameworks/libboost_thread-mt.dylib
       0x10dc3c000 -        0x10dc40fff +libboost_chrono-mt.dylib (0) <A19D3675-D6DF-3AE5-8271-5F346D13E5B4> /Applications/Scopy.app/Contents/Frameworks/libboost_chrono-mt.dylib
       0x10dc4b000 -        0x10dc4ffff +libboost_date_time-mt.dylib (0) <888222D0-8C27-3E38-8F7A-C9D52CEF661C> /Applications/Scopy.app/Contents/Frameworks/libboost_date_time-mt.dylib
       0x10dc65000 -        0x10dc66fff +libboost_atomic-mt.dylib (0) <D8FF0A88-D2C8-33BF-B5D4-F5DA292D60EA> /Applications/Scopy.app/Contents/Frameworks/libboost_atomic-mt.dylib
       0x10dc6c000 -        0x10dd49ff7 +libqwt.dylib (0) <1EA219BC-05F6-3A77-8E5F-C8D62B375A68> /Applications/Scopy.app/Contents/Frameworks/libqwt.dylib
       0x10ddbe000 -        0x10dddafff +libqwtpolar.1.dylib (0) <A5D69827-578C-3C0C-AE50-D7AFCE27F0BB> /Applications/Scopy.app/Contents/Frameworks/libqwtpolar.1.dylib
       0x10ddf7000 -        0x10de2afff +org.qt-project.QtSvg (5.13 - 5.13.0) <0BC7EC1B-367B-356F-9E9A-C05BBE103925> /Applications/Scopy.app/Contents/Frameworks/QtSvg.framework/Versions/5/QtSvg
       0x10de48000 -        0x10de78ff7 +org.qt-project.QtXml (5.13 - 5.13.0) <00BA31D1-B963-3402-BF9A-0D1D772D1D60> /Applications/Scopy.app/Contents/Frameworks/QtXml.framework/Versions/5/QtXml
       0x10de8e000 -        0x10de9dfff + (??? - ???) <4BA45C8B-11B1-3FF8-AF97-84EE08E48A30> /Applications/Scopy.app/Contents/Frameworks/iio.framework/Versions/0.18/iio
       0x10deab000 -        0x10defbfff +libmatio.9.dylib (0) <C7317EEF-24AC-3E48-8C75-97C30ACD1F1E> /Applications/Scopy.app/Contents/Frameworks/libmatio.9.dylib
       0x10df08000 -        0x10df88ff7 +libsigrok.4.dylib (0) <E3AAEAE3-4725-31E7-9DE2-68597B77C941> /Applications/Scopy.app/Contents/Frameworks/libsigrok.4.dylib
       0x10dfd5000 -        0x10e0c3fff +libglib-2.0.0.dylib (0) <0B78AEB5-56DD-3645-BEA7-C10242A49831> /Applications/Scopy.app/Contents/Frameworks/libglib-2.0.0.dylib
       0x10e0e9000 -        0x10e0f1ffb +libintl.8.dylib (0) <CD236902-DB50-39BB-86BB-E9F0D22E5649> /Applications/Scopy.app/Contents/Frameworks/libintl.8.dylib
       0x10e0f9000 -        0x10e11bfff +libsigrokcxx.4.dylib (0) <D8BFD2B2-65A6-3CB8-BAA2-81F5BC5C983C> /Applications/Scopy.app/Contents/Frameworks/libsigrokcxx.4.dylib
       0x10e166000 -        0x10e197fff +libglibmm-2.4.1.dylib (0) <27B49790-8A40-3A67-8E4A-10F001769111> /Applications/Scopy.app/Contents/Frameworks/libglibmm-2.4.1.dylib
       0x10e1ea000 -        0x10e223ffb +libgobject-2.0.0.dylib (0) <860193F3-1298-3EEF-B153-2B6D30F3865F> /Applications/Scopy.app/Contents/Frameworks/libgobject-2.0.0.dylib
       0x10e24e000 -        0x10e250fff +libsigc-2.0.0.dylib (0) <512571AF-4A7B-31BB-B302-9BDD1D43E38E> /Applications/Scopy.app/Contents/Frameworks/libsigc-2.0.0.dylib
       0x10e259000 -        0x10e260fff +libsigrokdecode.3.dylib (0) <A18F4691-0802-3DA1-B4E1-1EED8B595D17> /Applications/Scopy.app/Contents/Frameworks/libsigrokdecode.3.dylib
       0x10e26a000 -        0x10e375fff +org.qt-project.QtNetwork (5.13 - 5.13.0) <6C475B7F-8795-3FDE-9554-9891D0BDB84F> /Applications/Scopy.app/Contents/Frameworks/QtNetwork.framework/Versions/5/QtNetwork
[... 234 lines cut ...]
    0x7fff6c7d0000 -     0x7fff6c7d3ff7  libsystem_sandbox.dylib (851.270.1) <9494594B-5199-3186-82AB-5FF8BED6EE16> /usr/lib/system/libsystem_sandbox.dylib
    0x7fff6c7d4000 -     0x7fff6c7d6ff3  libsystem_secinit.dylib (30.260.2) <EF1EA47B-7B22-35E8-BD9B-F7003DCB96AE> /usr/lib/system/libsystem_secinit.dylib
    0x7fff6c7d7000 -     0x7fff6c7deff3  libsystem_symptoms.dylib (820.267.1) <03F1C2DD-0F5A-3D9D-88F6-B26C0F94EB52> /usr/lib/system/libsystem_symptoms.dylib
    0x7fff6c7df000 -     0x7fff6c7f4fff  libsystem_trace.dylib (906.260.1) <FC761C3B-5434-3A52-912D-F1B15FAA8EB2> /usr/lib/system/libsystem_trace.dylib
    0x7fff6c7f6000 -     0x7fff6c7fbffb  libunwind.dylib (35.4) <24A97A67-F017-3CFC-B0D0-6BD0224B1336> /usr/lib/system/libunwind.dylib
    0x7fff6c7fc000 -     0x7fff6c82bfff  libxpc.dylib (1336.261.2) <7DEE2300-6D8E-3C00-9C63-E3E80D56B0C4> /usr/lib/system/libxpc.dylib

Model: MacBookPro11,1, BootROM 156.0.0.0.0, 2 processors, Intel Core i5, 2,6 GHz, 8 GB, SMC 2.16f68
Graphics: kHW_IntelIrisItem, Intel Iris, spdisplays_builtin
[... cut ...]
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x112), Broadcom BCM43xx 1.0 (7.77.61.2 AirPortDriverBrcmNIC-1305.8)
Bluetooth: Version 6.0.14d3, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256F, 251 GB
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: USB 10/100/1000 LAN
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB Optical Mouse
USB Device: USB2.0 Hub
USB Device: USB Multimedia Keyboard
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.2

AlexandraTrifan, 2019-09-19:
Hello,

Can you try this new package and confirm that it works?

https://we.tl/t-13zlekzLll

If everything is fine, we will create a new release with the patch as soon as possible.

Thank you.
-Alexandra

todorspetkov, 2019-09-19:
Hi, I confirm that this new package resolves the missing libicui18n
issue on my Mac.
Thanks a lot,
    Todor Petkov

amiclaus, 2020-01-16:
The bug fix was confirmed. The issue will be closed.

closed 2020-01-16 by amiclaus