analogdevicesinc/scopy #712: General: "Calibration failed" after re connecting the board

andreeaipop, 2019-09-19:
OS Version: Windows 10 (10.0)
FW Version: v 0.24
SW Version: 6fa2c03

Description of the bug: "Calibration failed " message appears if you disconnect and re-connect the board then press Calibrate.
Steps to reproduce:
Open Scopy, connect m2k, click Calibrate.
Click Disconnect, then Connect again. Click Calibrate and there will appear "Calibration Failed" message.
[image]

If you Disconnect, the message will disappear but after clicking  on Calibrate the message will appear again. If you repeat this 3-4 times, after clicking Disconnect the names of the tools which were calibrated freeze as "Calibrating..."
[image]

At this point, if you connect the m2k again the tools will work but the names are still replaced with "Calibrating...". The names remain like this even if you unplug the board from the USB port.
[image]

The ini files might be useful to reproduce the error.

[Scopy.ini.zip](https://github.com/analogdevicesinc/scopy/files/3630999/Scopy.ini.zip)

AlexandraTrifan, 2019-10-07:
This should be fixed on the latest master b677caed47d5d32ba3ceda4eb25438e36bff226f

andreeaipop, 2020-01-16:
In Scopy v1.1.2 the bug is fixed

closed 2020-01-16 by andreeaipop