analogdevicesinc/scopy #714: Pattern Generator: changing parameters of a channel affects the signal generated on another channel

andreeaipop, 2019-09-20:
OS Version: Windows 10 (10.0)
FW Version: v 0.24
SW Version: 6fa2c03

**Describe the bug**
Changing the settings of a high frequency signal in pattern generator affects the signal generated on another channel (at a significant smaller frequency value).

Steps to reproduce the behavior:
Open Scopy Pattern generator and enable 2 random digital channels, one with low frequency and one with high frequency.
 Configure one channel ( DIO6 in this case) to 50 Hz frequency and 50% Duty cycle.
Configure the second channel (DIO8) to 1Mhz and 50% Duty cycle.
If you change the frequency of the second channel to certain values (3 Mhz, 4Mhz, 7 Mhz, 16Mhz ...) pattern generator automatically adjuts the duty cycle of the channel, but also affects the signal generated on the lower frequency channel.

**Expected behavior**
Changing the parameters of one digital channel should not affect the signal on another digital channel.

**Screenshots**
This is how th 50Hz signal (DIO6) and 1 Mhz signal (DIO8) look in logic analyzer:
[image]
Change the frequency of DIO8 to 3 Mhz:
[image]

The signals will look like this even if nothing was changed from the settngs of DIO6
[image]

DanielGuramulta, 2021-01-11:
Issue mentioned in #949 solved with #971

tagoylo, 2021-04-07:
Issue still exists on Scopy v1.3 build 441fd193.
[image]
[image]

adisuciu, 2021-04-07:
This cannot be fixed due to the way hardware works. We put a warning in the pattern generator when this happens - "Maximum buffer size reached"
-Adrian

tagoylo, 2021-04-07:
Got it. When the warning occurs, this is expected. Will close this item.

closed 2021-04-07 by tagoylo