analogdevicesinc/scopy #721: ADALMA2000 : scopy window too large

ecy77, 2019-10-28:
**Environment:**
 - OS: Debian Linux 64 bits
 - FW Version 0.24 
 - SW Version 1.1.1

**Describe the bug**
When I run the flatpak Scopy, his window is too big for my screen resolution (1920x1080). I can't resize the window to fit my screen and font is too big.

**To Reproduce**
Steps to reproduce the behavior:
1. Run <flatpak run org.adi.Scopy>

**Expected behavior**
I was expecting a window that takes the size of my screen.

**Screenshots**
[image]

**Additional context**

I use Gnome with wayland. Disable wayland don't change anything

cxcorp, 2019-11-08:
I'm having the same issue on Ubuntu 18.04 on v1.1.1. The UI works as expected on v1.0.6, so I suspect this is a regression caused by #663.

DanielGuramulta, 2019-11-08:
Hi,
Unfortunately I was unable to reproduce the issue with the screen resolution set to `1920x1080`, but, when using `3840x2160` the issue was as described. After some investigation, I agree with you that the issue might be caused by #663.

I will take a closer look into this issue and will try to find a fix for this.

DanielGuramulta, 2019-11-08:
Hi again,
I had a closer look at the issue, and it seems that the fix for high dpi displays in #663 is not compatible on Linux, until we find a solution for that, disabling what was implemented in #663 for Linux platforms solved the issue for me. I have uploaded here https://we.tl/t-sZrkAtV828 a new flatpak installer with the fix. Can you give it a try if you have some time and confirm that the issue is also fixed for your system.

Thank you,
Daniel

cxcorp, 2019-11-08:
Hey Daniel,

Thanks for the superbly quick response! I just gave that dev build a go - the issue appears fixed on my system!

ecy77, 2019-11-19:
Sorry for the delay.
Everything is ok with this new flatpak
thanks

closed 2019-11-19 by ecy77

pt1270, 2019-12-10:
Hi,
I have exactly the same issue on elementary OS 5.1 Hera. Unfortunately i wasn't able to download the installer Daniel provided.
Is there the possibility to get the fixed flatpack? 

Best Regards,
Philipp

pt1270, 2019-12-10:
> Hi again,
> I had a closer look at the issue, and it seems that the fix for high dpi displays in #663 is not compatible on Linux, until we find a solution for that, disabling what was implemented in [...]

Hi,
I have exactly the same issue on elementary OS 5.1 Hera. Unfortunately i wasn't able to download the installer Daniel provided.
Is there the possibility to get the fixed flatpack?

Best Regards,
Philipp