analogdevicesinc/scopy #722: Scopy 1.1.1 not compatible with macOS 10.13?

bobgates, 2019-10-31:
**Environment:**
 - OS: macOS 10.13 on MacBook Pro 2017

**Describe the bug**
Can't install: 
[image]

On the website, the install file is called Scopy-v1.1.1-osx_10.13.dmg

However, on attempting to install it, I am seeing the above error.

AlexandraTrifan, 2019-10-31:
Hi, 

We are looking into this, it might be a packaging mistake. We will come back as soon as possible with a response/fix.

-Alexandra

AlexandraTrifan, 2019-11-19:
Hi,

I created another dmg. Can you try this one?
https://transfer.sh/BUsF0/Scopy-v1.1.1-osx_10.13-patch.dmg.zip

-Alexandra

bobgates, 2019-11-19:
Runs on 10.13 now. Thanks.

closed 2019-11-19 by bobgates