analogdevicesinc/scopy #723: Scopy v1.0.4: Can't see ADALM2k on USB on Mac

bobgates, 2019-10-31:
**Environment:**
 - OS: macOS 10.13 on 2018 Macbook Pro

**Describe the bug**
I installed Scopy in /Applications. It starts up and runs okay but no ADALM2000 module appears available. I have reinstalled the HoRNDIS driver and rebooted. I've tried to find useful info in the logs, but I'm not seeing anything with the word HoRNDIS which worries me a little.

**Expected behavior**
I expected Scopy to start up, see my ADALM,  _then_ crash in some bizarre way a bit later. Forgive me, I've found using it on a Mac a little trying.

Any help appreciated.

bobgates, 2019-10-31:
> Reply to: analogdevicesinc/scopy <reply@reply.github.com>
>
> Hi,
>
> Can you see M2k (or ADALM2000) as a mass storage device in Finder?
>
> Thank you,
> -Alexandra

Yes.

bobgates, 2019-10-31:
Oops – the correct answer is No. What I was seeing was one of the install packages.

I’m not seeing the ADALM2k in Finder.

bobgates, 2019-10-31:
Further news – started to suspect the cable. Indeed – a new USB cable means I can now see the device and Scopy is working. Aaarghhh. It’s always annoying when it turns out to be hardware!

Apologies for wasting your time. It would still be good to get 1.1.1 on macOS 10.13 though.

AlexandraTrifan, 2019-11-01:
Hello,

We will look into the 10.13 issue. Meanwhile, I am closing this issue.

Thank you,

-Alexandra

closed 2019-11-01 by AlexandraTrifan