analogdevicesinc/scopy #732: Voltmeter: Data logging function does not work on v1.1.1

keshikan, 2019-11-26:
**Environment:**
 - OS Version: Windows 10 (1903)
 - FW Version: v0.25
 - SW Version: v1.1.1

**Describe the bug**
Hello, support.
Data logging function does not work on v1.1.1, but works correctly on alpha v0.11 .

**To Reproduce**
1. Start Scopy v1.1.1, and connect to the device.
1. Click "Run" button on Voltmeter tab.
2. Click "Browse" button and select a *.csv file.
3. Click "ON" button to start logging
4. ON button can't be toggled, and"No file selected" is displayed.

[image]

Thank you.

adisuciu, 2019-12-03:
Fixed with https://github.com/analogdevicesinc/scopy/pull/734

amiclaus, 2020-01-16:
Scopy Version 1.1.2
The bug is fixed.

closed 2020-01-16 by amiclaus