analogdevicesinc/scopy #737: Scopy v1.1.1 crashes immediately after connection attempt

Kabron287, 2019-12-10:
**Environment:**
 - OS: Windows 10 LTSC x64
 - FW Version 0.25
 - SW Version 1.1.1

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Just try to Connect

adisuciu, 2019-12-11:
Did you install the driver for Windows ? - https://wiki.analog.com/university/tools/pluto/drivers/windows

Make sure you restart the computer after installing. If that doesn't work, can you connect through an ip context ? https://wiki.analog.com/university/tools/m2k/scopy#connecting_to_a_remote_device
Ip address should be 192.168.2.1

-Adrian

Kabron287, 2019-12-11:
No, attempt to connect via IP also leads to crash.
At the same time Scopy 1.1.0 works fine.

adisuciu, 2019-12-11:
Can you try this build - https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/29249927/job/90nnoc5h6nu1jvv7/artifacts ?

It should have fixed some of the crashes.

-Adrian

Kabron287, 2019-12-11:
Hello, 
file, you suggest(https://ci.appveyor.com/api/buildjobs/90nnoc5h6nu1jvv7/artifacts/scopy-64-setup.exe) is byte-to-byte equal  with my Scopy 1.1.1 scopy-64-setup.exe.
BTW, Scopy 1.1.0 also crashes with IP access.

adisuciu, 2019-12-11:
scopy-v1.1.1-Windows-setup.exe from the release page has 176 MB as it contains both 32 and 64 bit versions - https://github.com/analogdevicesinc/scopy/releases

The file I suggest is 90.4MB and it should've fixed some stuff from the build server since the last release.

Can you give me the git-tag of the file that crashes. It's in the title bar.
-Adrian

Kabron287, 2019-12-12:
OK,
Scopy-v1.1.1 6f32056  from  scopy-v1.1.1-Windows-setup.exe(184MB) works like a charm.
Scopy-v1.1.1 c6e5be9 (the subject of the issue) now also works fine.
I gues new setup corrects something. 
Thanks and sorry for disturbance

closed 2019-12-12 by adisuciu