analogdevicesinc/scopy #738: ADALMA2000: Scopy window too big for Display, can't be adjusted smaller

pt1270, 2019-12-10:
**Environment:**
 - OS: elementary: OS 5.1 Hera 
 - FW Version: 0.24
 - SW Version: V1.1.1
 - Screen Resolution: 1920x1080

**Describe the bug**
When opening Scopy (installed via flatpack) the window is way too large for my screen. I am not able  to make it smaller.

**To Reproduce**
Steps to reproduce the behavior:
1. Run Scopy V1.1.1 on elementary OS 5.1 

**Expected behavior**
Window fitting the screen. 

**Screenshots**
[image]

**Additional context**
I think  [#721](https://github.com/analogdevicesinc/scopy/issues/721) is the same problem as mine but im not able to download the fixed flatpack installer since the link is expired.

DanielGuramulta, 2019-12-11:
Hi,

We already have the fix merged on master. We will make a new release this week, where you would be able to get the new flatpack installer.

adisuciu, 2020-02-12:
Should be fixed.

closed 2020-07-16 by adisuciu