analogdevicesinc/scopy #743: Problems building Scopy: breakpad_client

bobgates, 2019-12-18:
Thanks for the help patching qwt.

I'm slowly getting the elements of Scopy to build in MacOs 10.14, and have reached the point at which all the dependencies on the Wiki page are compiled and I can actually start to build Scopy. First question: how can I contribute to the wiki page:
https://wiki.analog.com/university/tools/m2k/scopy/build-osx#installing_the_dependencies
I hit a few speed bumps that I'd like to document for anyone else who is interested.

Second question: the cmake command in the Scopy directory came back with two things missing: Breakpad and Matio. Breakpad and Matio installed easily enough, but cmake is now reporting: BREAKPAD_CLIENT_LIBRARIES as linked by Scopy but not found. The file it's looking for seems to be breakpad_client, and a search of my hard-disk doesn't find it.

Can you help with a lead? I feel like I'm really close...

Thanks,
Declan

adisuciu, 2019-12-19:
Hi,
Take a look at the commit in this pull request to disable building breakpad on your machine
https://github.com/analogdevicesinc/scopy/pull/744

-Adrian

bobgates, 2019-12-20:
This commit solves the breakpad halting of the build, thanks.

closed 2019-12-20 by bobgates