analogdevicesinc/scopy #747: Problem building Scopy: intl

bobgates, 2019-12-20:
**Environment:**
macOs 10.13 on 2017 Macbook Pro

I'm trying to build Scopy for the Mac, following the instructions on:
[https://github.com/analogdevicesinc/scopy/wiki/Building-Scopy-on-OS-X](https://github.com/analogdevicesinc/scopy/wiki/Building-Scopy-on-OS-X )

**Semi-resolved issues**
There was a library needed (pkg-config) that wasn't specified in the instructions,but homebrew looked after, but I found it and all good.

There was also a fail in compiling network_analyzer.cpp - two calls to:
goertzel1->set_len(buffer_size);
failed because set_len is not defined for goertzel. I've commented the two lines out, don't yet know what the results of that will be.

**Bug**
My current roadblock is during linking of Scopy:
ld: library not found for -lintl
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [Scopy.app/Contents/MacOS/Scopy] Error 1

I'm at a loss where to find the intl library, and why it isn't available to the Linker.

All help would be appreciated.

adisuciu, 2019-12-20:
Hello,
Are you using our branch of gnuradio ? You are supposed to use the scopy branch from our repository. This branch has the set_len method implemented.

https://github.com/analogdevicesinc/gnuradio/tree/scopy

According to the build guide on the wiki:
```
cd ~
git clone https://github.com/analogdevicesinc/gnuradio
cd gnuradio
git checkout scopy
mkdir build && cd build
cmake -DENABLE_INTERNAL_VOLK:​BOOL=OFF -DENABLE_GR_FEC:BOOL=OFF -DENABLE_GR_DIGITAL:BOOL=OFF -DENABLE_GR_DTV:BOOL=OFF -DENABLE_GR_ATSC:BOOL=OFF -DENABLE_GR_AUDIO:BOOL=OFF -DENABLE_GR_CHANNELS:BOOL=OFF -DENABLE_GR_NOAA:BOOL=OFF -DENABLE_GR_PAGER:​BOOL=OFF -DENABLE_GR_TRELLIS:​BOOL=OFF -DENABLE_GR_VOCODER:​BOOL=OFF ​..
make
sudo make install
```

-Adrian

adisuciu, 2019-12-20:
lintl package is part of the gettext package. You are supposed to install that as well
brew install gettext
We will update our build guide.
-Adrian

bobgates, 2019-12-20:
Yeah, please do! (update the guide)

I don't know if I installed gnuradio from your build, or perhaps just missed the git checkout scopy line. I'm busy reinstalling. Does that mean I should rebuild gr-iio as well? Because that linked okay, and I think it might have linked to a copy of gnuradio sitting somewhere else on my directory tree.

adisuciu, 2019-12-20:
Yes, you should rebuild gr-iio.

bobgates, 2019-12-20:
I've installed gettext with brew, but still got the same linker error.

It turns out this has happened before: https://github.com/EOSIO/eos/issues/2028

I'm not sure how to read that, since I just installed gettext today, but somehow it didn't get linked properly. Anyway: 
 `brew unlink gettext && brew link --force gettext`
is the solution.

I've also rebuilt everything from gnuradio onwards to get rid of old copies of things, I'm now running on the scopy branch, and that's working. I like the look of 1.1.1 - it seems a little ahead of whatever I was running!

bobgates, 2019-12-29:
Note that the MacOs guide, https://wiki.analog.com/university/tools/m2k/scopy/build-osx, differs from the advice given by Adrian, above. Adrian has:

cd ~
git clone https://github.com/analogdevicesinc/gnuradio
cd gnuradio
git checkout scopy

whereas the MacOs guide has:
cd ~
git clone https://github.com/analogdevicesinc/gnuradio
cd gnuradio
git checkout signal_source_phase

That took some finding. Please can you update the guide?

Thanks,
Declan

adisuciu, 2020-01-07:
Updated ! Thanks !

closed 2020-01-07 by adisuciu