analogdevicesinc/scopy #750: Logic Analyzer: SPI decoder problem (Mode 1)

TKTF50, 2020-01-12:
**Environment:**
    OS: Windows 10
    FW v0.24
    SW V1.1.1

**Describe the bug**
In the Logic Analyzer, the values shown in the group interpretation of SPI data are incorrect for Mode 1 (CPOL = 0, CPHA = 1).

**To Reproduce**
Steps to reproduce the behavior:
1. Logic Analyzer instrument
2. SPI decoder (without using MISO signal)
3. CS# polarity: active-low
4. Clock polarity: 0
5. Clock phase: 1
6. Bit order: msb-first
7. Word size: 8
8. Capture trace

Result >> _SPI: MOSI data_ and _SPI: MOSI bits_ are incorrect.

**Expected behavior**
It's my understanding that for CPOL=0, CPHA=1, the slave looks at the MOSI signal on the falling clock edge (1 to 0 transition). I have attached a screenshot (Scopy Logic Analyzer 1.png) that shows the leftmost cursor over the 4th falling clock edge. MOSI is 1 at this time and yet the MOSI bits show a 0 here. It looks like the MOSI signal is being sampled on the rising clock edge rather than the falling one.

**Screenshots**
[image]
[image]

[Scopy.ini.txt](https://github.com/analogdevicesinc/scopy/files/4051302/Scopy.ini.txt)

**Additional context**

The trace was captured with an Arduino Uno communicating with a TI DAC8552. I know the correct values are actually being transmitted because the DAC responds correctly (in the "Scopy Logic Analyzer 1" screenshot, the value should be 0x10 rather than 0x08). The full data capture is shown in screenshot "Scopy LA Full.png" (which also shows the General Settings in use for the test case). The 3 bytes of MOSI data should be shown as: 10 34 3E.

I can't reproduce this with just the Pattern Generator and Logic Analyzer. (Not sure it matters, but I can't get the DAC8552 to respond if I send it data directly from the Pattern Generator - but maybe I'm doing it wrong since it is the first time I've used the Generator.)

adisuciu, 2020-01-13:
<scopy_installation>\decoders\spi\pd.py contains the implementation of the SPI protocol decoder from libsigrokdecode. 
Can you edit the file and try changing lines 67-72 from

TKTF50, 2020-01-13:
I made the change you suggested and then ran the Analyzer for multiple repeats of the same command (10 33 33). The decoded SPI data changes  across runs (see attached screenshots). Sometimes it is correct, sometime wrong and when it's wrong there are a variety of "off by one bit" problems. This makes me think it is still interpreting MOSI on the rising clock edge.

[image]
[image]

TKTF50, 2020-01-13:
I switched back to the original pd.py code and ran the same test case (i.e. data should be 10 33 33). In this case, the result was consistently 88 19 99 (no change between runs like there was with the code change in place). Hope that helps with the analysis.

adisuciu, 2020-10-23:
We created a new logic analyzer. This should have the latest sigrok libraries for decoding signals. Maybe you can check it out to see if the issue with the SPI mode was resolved.
-Adrian

TKTF50, 2020-10-23:
@adisuciu Thanks for the update. What version would you like me to check? Is the new logic analyzer in release 1.1.2 or the nightly build?

adisuciu, 2020-10-30:
Yes, we released this version today which fixes more stuff from RC1.

adisuciu, 2020-10-30:
Yes we released this version You can test it here https://github.com/analogdevicesinc/scopy/releases/tag/v1.2.0 which fixes some stuff from RC1

adisuciu, 2020-10-30:
Alright, I tried reproducing, I tried reproducing by sending a SPI message from the pattern generator. 

I'm actually missing some bits of the beginning of the signal (which might mean that you are receiving only the last command, not the first). Upon further investigation it turns out that this was due to a trigger delay setting which was not set properly in logic analyser streaming mode. As soon as I changed to oneshot, the delay control appeared and I could set the delay properly and get both of the messages.

1st screenshot - data missing - decoding bad - 
[image]

2nd screenshot - data ok - after correctly setting the delay.

[image]

Can you confirm that this is what happened to you too ?

TKTF50, 2020-10-30:
My mistake Adrian. I had enabled some debug messages in my code so that I could see the bytes that were supposed to be sent. The debug messages stretched the time between the SPI messages from 25us to 5.8ms! Doh!

When I disabled the debug messages, I could see both expected SPI messages in the logic analyzer. I'm closing this issue.

Thanks!

closed 2020-10-30 by TKTF50