analogdevicesinc/scopy #770: Signal generator channel synchronization Bug

damercer, 2020-03-02:
**Environment:**
 - OS: Win 10
 - FW Version 0.25
 - SW 1.1.2 release

**Describe the bug**
Signal generator channel waveform outputs loose synchronization after setting waveform to Trapezoidal. Does not always come un-locked but if the correct steps are followed (see bellow) then it seems to be reproducible.

**To Reproduce**
Steps to reproduce the behavior:
In the following example. In the Signal generator set both channels the same. Set the Frequency to 25kHz, Amplitude to 2 V (any value will do), Offset to 2.5V  (any value will do), Phase to 0. Set to Sine waveform.
With signal generator outputs looped back to scope inputs. 
Run Signal Generator.
Run Scope. Set channel range, offset and time base to observe waveforms.
Set to trigger on channel 1. Both scope traces will be (should be) on top of each other, that is synced.
(first screen shot)
Now back in the Signal Generator. Change channel 2 to Trapezoidal with rise/fall time set to 1uS and high / low time set to 19uS. Frequency should be 25 kHz.
Go back to Scope. Waves are now sliding across each other ( see second screen shot hard to show sliding in still picture)
Back in Signal Generator set shape back to sine and wave forms still not sync'd (see third screen shot)
Now type 25 (kHz) in for Frequency  hit return. Wave forms back in sync. See fourth screen shot.

**Expected behavior**
Waveforms should always remain in sync at the set relative phase.

**Screenshots**
Waveforms are in sync.
[image]

Wave forms are sliding across each other, not sync'd.
[image]

Switch back to sine shape and wave forms still not sync'd
[image]

After entering Frequency as 25 kHz again. Back to as expected.
[image]

**Additional context**

The ini files might be useful to reproduce the error.
[Scopy.zip](https://github.com/analogdevicesinc/scopy/files/4275714/Scopy.zip)

adisuciu, 2020-04-23:
https://github.com/analogdevicesinc/scopy/pull/782 should fix it

HRosete, 2020-10-21:
This bug is now fixed at Scopy v1.2.0 (verified at 33ca72f9). Closing this item.

closed 2020-10-21 by HRosete