analogdevicesinc/scopy #775: Scopy requires sudo to detect ADALM2000

etofigh, 2020-04-07:
After following the installation guide for linux, executing `flatpak run org.adi.Scopy` opens the Scopy window but no device is detected. Even with `flatpak run --device=all org.adi.Scopy` it does not work. When executing the `sudo flatpak run org.adi.Scopy` the device is correctly detected.

adisuciu, 2020-04-07:
Did you install the udev rules for the device ? https://wiki.analog.com/university/tools/pluto/drivers/linux

> In order to access some USB functions without root privileges, it's recommended to install the PlutoSDR or ADALM2000 udev rules. Simply download 53-adi-plutosdr-usb.rules or 53-adi-m2k-usb.rules and copy into into the /etc/udev/rules.d/ folder. You can also use this plutosdr-m2k-udev.deb package to install both the PlutoSDR and the ADALM2000 udev rules. To install the debian package, use one of the following: sudo apt-get install ./plutosdr-m2k-udev.deb or sudo dpkg -i plutosdr-m2k-udev.deb.
> Afterwards reload rules or restart udev, using either udevadm control --reload-rules or sudo service udev restart

etofigh, 2020-04-09:
Thank you for the help! I copied the `53-adi-m2k-usb.rules` file manually to the `/etc/udev/rules.d`. Reloaded the rules but it didn't work. But after adding my user to the `plugdev` group and restarting the OS it worked without sudo!
Thanks again!

closed 2020-04-09 by etofigh