analogdevicesinc/scopy #776: win10 msys2 x64 Compiler error

MEGA-Leo, 2020-04-09:
**Environment:**
 - OS: [win10 X64]
 - 1909(18363.752)
 - https://github.com/analogdevicesinc/scopy.git

**Describe the bug**
just do it follow https://wiki.analog.com/university/tools/m2k/scopy/build-windows#building_scopy1,but error as follow

  C:/msys64/mingw64/lib/cmake/gnuradio/gnuradio-fftConfig.cmake:24 (find_dependency)
  C:/msys64/mingw64/lib/cmake/gnuradio/GnuradioConfig.cmake:116 (include)
  CMakeLists.txt:115 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:118 (find_package):
  Could not find a package configuration file provided by "gnuradio-iio" with
  any of the following names:

    gnuradio-iioConfig.cmake
    gnuradio-iio-config.cmake

  Add the installation prefix of "gnuradio-iio" to CMAKE_PREFIX_PATH or set
  "gnuradio-iio_DIR" to a directory containing one of the above files.  If
  "gnuradio-iio" provides a separate development package or SDK, be sure it
  has been installed.

I have download scopy-mingw64-build-deps.tar.xz and overwirte to C:\msys64\

Please help me,Thanks!!

adisuciu, 2020-04-10:
The build guide is outdated unfortunately due to the upgrade to the new versions of the gnuradio 3.8 library, gr-iio, etc. 

Meanwhile, you can use this guide with an older branch of scopy - git checkout v1.1.2 - for example
Or you can also take a look in https://github.com/analogdevicesinc/scopy/blob/master/CI/appveyor/build_appveyor_mingw.sh

This shell script builds Scopy on our appveyor CI on Windows. It's similar to the environment you have on your Windows machine.

We will update the documentation on how to build Scopy on the wiki as soon as possible

-Adrian

adisuciu, 2020-05-22:
There's a new build guide available on the wiki: https://wiki.analog.com/university/tools/m2k/scopy/build-windows?s[]=scopy&s[]=build

closed 2020-05-22 by adisuciu