analogdevicesinc/scopy #779: <Instrument>: <Short description of the bug>

ciuffoly, 2020-04-21:
OS Version: Windows 10 (10.0)
FW Version: v0.25
SW Version: 3d4c582

Description of the bug: hang
Steps to reproduce:
I have used the ADALM2000   yesterday without problems
This morning any time I start the Oscilloscope on Scopy the program hang (it is locked with Not responding)
I have update now the firmware, but was the same release
I have reboot the PC
I have reinstalled the Scopy software

What to do now ?

DanielGuramulta, 2020-04-21:
Hello

Could you please upload the Scopy.ini file. It should be located in this path:
`C:/Users/<username>/AppData/Roaming/Scopy.ini`
After uploading it could you also try deleting it and then start Scopy again and check if the issue persists.

Daniel

ciuffoly, 2020-04-21:
I don't find this fine here

C:\Users\ciuffoly\AppData\Roaming

ciuffoly, 2020-04-21:
Found here

C:\Users\ciuffoly\AppData\Roaming\ADI[

](url)

DanielGuramulta, 2020-04-21:
`C:/Users/<username>/AppData/Roaming/ADI/Scopy.ini` maybe in the ADI folder, I got the path wrong

ciuffoly, 2020-04-21:
yes found
[Scopy.zip](https://github.com/analogdevicesinc/scopy/files/4507963/Scopy.zip)

DanielGuramulta, 2020-04-21:
Thank you!

Will try to see if I can reproduce the bug. Did Scopy work after deleting this file?

ciuffoly, 2020-04-21:
never tested, I need to try this delete ?

DanielGuramulta, 2020-04-21:
Yes, you could try to delete this file while Scopy is not running and then try to run Scopy again

ciuffoly, 2020-04-21:
good I have rename this file and now all work !

thank you 

on this page the Installer for latest (nightly) build (Windows 64-bit) is not a godo link
https://wiki.analog.com/university/tools/m2k/scopy
is there a release after 1.1.2 ?

DanielGuramulta, 2020-04-21:
You can get the installer from here: https://github.com/analogdevicesinc/scopy/releases
v1.1.2 is the latest. Unfortunately the links on the wiki for the installers don't work anymore and also that build guide needs to be updated. Thanks for also reporting this problems

Daniel

ciuffoly, 2020-04-21:
thank you very much for your support

ciuffoly, 2020-04-21:
my webpage about this instrument
http://www.audiodesignguide.com/AudioMesurement/index.html

ciuffoly, 2020-04-21:
do you think will be possible add some more functions:

1) modify the signal generator level in the Oscilloscope panel
2) add Vrms level in the Oscilloscope panel
3) show Slewrate

DanielGuramulta, 2020-04-21:
Will open another issue regarding your questions of the features to be implemented and will close this one regarding the Scopy hang.

Daniel

closed 2020-04-21 by DanielGuramulta

adisuciu, 2020-04-21:
@ciuffoly replied in the new thread