analogdevicesinc/scopy #780: do you think will be possible add some more functions:

DanielGuramulta, 2020-04-21:
do you think will be possible add some more functions:

1) modify the signal generator level in the Oscilloscope panel
2) add Vrms level in the Oscilloscope panel
3) show Slewrate

_Originally posted by @ciuffoly in https://github.com/analogdevicesinc/scopy/issues/779#issuecomment-616977203_

adisuciu, 2020-04-21:
1. All tools can be detached by dragging them out of the left Tool Launcher onto the application or by double clicking the instruments if that setting is set in Preferences - https://wiki.analog.com/university/tools/m2k/scopy?s[]=detach#user_guides

2. VRMS is available in measurements -https://wiki.analog.com/university/tools/m2k/scopy/oscilloscope?s[]=measurements&s[]=scopy#using_built-in_signal_measurement_features
3. We don't have slewrate implemented but there is computation of rise and fall time.

-Adrian

ciuffoly, 2020-04-21:
Very good, it work so point 1 and 2 are solved!

3 is not important

ciuffoly, 2020-04-21:
Is possible to see thd level on oscilloscope ?

adisuciu, 2020-04-21:
Unfortunately there are not many spectral measurements implemented right now - we have a separate issue here https://github.com/analogdevicesinc/scopy/issues/526 but no ETA.
-Adrian

closed 2020-04-27 by adisuciu