analogdevicesinc/scopy #788: Scope: Phase adjustment in math channel OR display

dsandber, 2020-05-15:
I'm trying to graph the three phases of a three-phase motor.  I figured I'd do the third phase with a math-channel, offsetting the phase of the second winding by 120 degrees.

However, it seems (please correct me if I missed something) that there's no way to do phase offsets.  Ideally I could specify it when creating the math channel, but I didn't see a way.

Alternatively, I figured I could shift a math channel that is a copy of the "t1" phase in the scope display (in the same way I can move a trace up and down) but there doesn't seem to be a way to move a single phase left and right.

As a work-around I could imagine using the signal generator, but it seems like that shouldn't be necessary.

How come individual traces can be shifted up and down but not left and right?

Relatedly, it's confusing that some controls are tied together, and some are separate, but nothing shows which is which -- for example, changing the voltage position will change only one channel, but changing the time position affects all traces.

damercer, 2020-05-19:
In a standard oscilloscope this sort of operation is not done, possible or allowed. The time axis is fixed relative to the triggering event for all input channels.

dsandber, 2020-05-19:
Oh, I must have misremembered, it's been a while since I used a physical scope.  Any idea why the voltage axis would be movable per trace but not the time axis?

Well, if the goal is not to deviate from a physical scope, then perhaps the calculation channels could have some way of specifying phase offset?

damercer, 2020-05-19:
If you want to view three waveforms then you would need a three channel (or more) oscilloscope. In the case of using a two channel scope like M2K and Scopy, you can connect channel 1 to one waveform and trigger on channel 1, connect channel 2 to the second waveform. Now take a snapshot of channel 2 (Ref1 trace) now move channel 2 connection to third waveform. Now you are "viewing" three waveforms but only two can be live i.e. real time, at a time. And relative to trigger point of channel 1.

closed 2020-07-16 by adisuciu