analogdevicesinc/scopy #798: scope: scopy crashes when recording captured video

balancedaction54, 2020-05-22:
**Environment:**
 - OS: Win 10
 - FW Version 0.22
 - SW Version or git tag: Scopy v1.1.2

**Describe the bug**
I can't use Scopy while also using software to record my desktop, because Scopy repeatedly crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Scopy and video capture software (Camtasia, OBS, or Zoom)
2. Turn on the oscilloscope
3. Try to change the time/div
4. Scopy crashes

[Scopy.ini.txt](https://github.com/analogdevicesinc/scopy/files/4670179/Scopy.ini.txt)

**Expected behavior**
I expected the scope to work as usual, so I can record a video lecture showing its use.

**Screenshots**
Not applicable, it does not matter what signal is being displayed.

**Additional context**

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/ali/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

Add any other context about the problem here.
I have tried three different video capture programs (Camtasia, OBS and Zoom) and three different computers (Win 10 laptop, Win 10 desktop, Win 7 laptop) all with the same result.
I attached the ini file with a .txt extension.

adisuciu, 2020-05-25:
Hello, 
It works just fine for me .. I used OBS to try and reproduce. I captured whole desktop instead of application

[2020-05-25_13-55-49.zip](https://github.com/analogdevicesinc/scopy/files/4676861/2020-05-25_13-55-49.zip)

Does it crash without recording ? Try removing the ini file and retry.

balancedaction54, 2020-05-26:
Hello, thank you for your feedback and help.  Unfortunately, I am unable to proceed as you show in the video you linked.  I tried recording the whole desktop instead of the program window or a selected portion of the desktop, and it did last a little longer before scopy crashed.  But I am unable to complete a fairly simple experiment that involves several steps of changing the input frequency and manipulating the scope controls, without scopy crashing.

It has crashed a few times without recording, and if it does so again I will do as you recommend and delete the .ini file.  But I am still unable to put together my lecture video as things stand.  If you have any other suggestions, please share them with me.

Thank you,
Steve Jacobs

closed 2020-07-08 by adisuciu