analogdevicesinc/scopy #807: Scopy::MingGW64 Build Issue

top58525785, 2020-06-18:
HI  I use MingGW to build Scopy

But when go to the last step, It Shows not find GWT_Cmake File

So how to set for QT?

****************************************
CMake Error at CMakeLists.txt:123 (find_package):
  By not providing "FindQwt.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qwt", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qwt" with any of
  the following names:

    QwtConfig.cmake
    qwt-config.cmake

  Add the installation prefix of "Qwt" to CMAKE_PREFIX_PATH or set "Qwt_DIR"
  to a directory containing one of the above files.  If "Qwt" provides a
  separate development package or SDK, be sure it has been installed.
*************************************************
Thanks Agiain
Jason.yu

adisuciu, 2020-06-18:
You didn't install the qwt dependency correctly 
https://wiki.analog.com/university/tools/m2k/scopy/build-windows?s[]=scopy&s[]=mingw#installing_the_dependencies

This wiki page shows how to build Scopy using QWT.

-Adrian

top58525785, 2020-06-19:
HI Adrian：
I install as what you said。
But still failed
：（

发自我的iPhone

 

 https://wiki.analog.com/university/tools/m2k/scopy/build-windows?s[]=scopy&amp;s[]=mingw#installing_the_dependencies
 
This wiki page shows how to build Scopy using MinGW.
 
-Adrian

top58525785, 2020-06-23:
HI Suciu：
I want to ask Scopy can build either using Mingw or QT creator？
Am i right？
Thanks again

发自我的iPhone

adisuciu, 2020-06-23:
QTcreator uses MinGW compiler to build. You can also build Scopy from the MSYS terminal.
https://wiki.analog.com/university/tools/m2k/scopy/build-windows?s[]=mingw#set_up_qtcreator

This chapter shows how to setup QtCreator to build Scopy.

-Adrian

top58525785, 2020-06-28:
HI Adrian:

How are you ?  Thanks for your support.

Now I also have some problems, I DO as what you told me,Using MinGW and QT creator.

Compile as bellows, but also failed:( I have do this for couple weeks. said...

Please help to check the attached error messages

Hope to receive you 

Best Regards
Jason.yu

测量与测试世界 www.vihome.com.cn
专注传感器，数据采集，电子测量仪器，系统集成，工业控制的专业媒体门户资讯和技术平台。

closed 2020-08-12 by adisuciu