analogdevicesinc/scopy #808: Scopy: Network Analyzer, option to keep offset between run iterations

benfre, 2020-06-20:
I try to use Network Analyzer to measure the response of a amplifier, which has very large DC gain and some input offset. I need to give the exciting generator a constant offset to prevent saturation of the amplifier. 

As between each frequency iteration the DAC are disabled, the amplifier output will ramping from saturation and back each time the offset are applied. 

Maybe a option to keep the offset (hold DAC to the offset voltage) can be added?

I can add a waveform of the amplifier output, if it helps understanding the situation.

benfre, 2020-06-20:
I think this will also helps for measuring response of transistor circuits.

adisuciu, 2020-06-22:
We're looking into it ..

benfre, 2020-06-22:
The reference wave:
[image]

The amplifier output:
[image]

damercer, 2020-06-25:
A simple solution might be to allow the user to specify a delay time for the AWG / external circuit to settle between frequency changes? Other vendor's tools allow this option. It will slow down the sweeping but can be set to zero when not needed for circuit settling.

Just a suggestion.

Doug

adisuciu, 2020-06-26:
This option is implemented - 
[image]

The naming is a little unfortunate, the first settling time represents the (minimum) time delay between AWG output and ADC acquisition. 

The second settling time (I think this should be renamed) - represents the (minimum) time delay between frequency changes (when the AWG is in high impedance)

The buffer previewer option lets you look into the ADC buffer in time domain, to check if everything looks good.

We're looking into the DAC holding the last sample and not going to high impedance, but this requires firmware/HDL changes. This would address the initial problem a little better.

-Adrian

benfre, 2020-06-26:
This may be a bug. The second settling time is not effective. When I increase it to 100ms, (showing in previous post pictures), the captured waveforms are still have saturation at the beginning (for other frequency not showing in picture).

Looking into the code, I guess this maybe due to retrieve waveform before the time delay.  I do not fully understand the underlying iio APIs, the line 1203 of `/src/network_analyzer.cpp` is refilling the buffer? and  line 1253 is the waiting.

benfre, 2020-06-26:
The two are named 'pushDelay' and 'captureDelay' in the code. 

But, I think the 'captureDelay' is not effective now. A bug?  

I don't know the iio functions , I only guess by there names. In my understanding the code, [/src/network_analyzer.cpp](../blob/master/src/network_analyzer.cpp):

- line 1145: wait 'pushDelay'
- line 1146-1149: enable DAC??
- line 1203: refill ADC buffer?? 
- line 1253: wait 'captureDelay'

So the second wait miss the point?

adisuciu, 2020-06-26:
Yes, you are correct, we changed the way the acquisition works, and we missed that. The fix is trivial, move the msleep function before refilling the buffer. Thanks for catching that. I can create a test build for you to test it out until we release the new software, if you are interested in that.

-Adrian

benfre, 2020-06-26:
> Yes, you are correct, we changed the way the acquisition works, and we missed that. The fix is trivial, move the msleep function before refilling the buffer. Thanks for catching that. I can create a [...]
Thanks. we can wait appveyor build, and try it.

adisuciu, 2020-07-29:
Did you get to try this ? 
-Adrian

benfre, 2020-08-08:
> Did you get to try this ?
> -Adrian

We test the fix. It works. Putting 500ms of settling time, the DC offset is low enough to enable high gain mode.

The reason we are facing this problem is because our amplifier board has a low pass frequency of few Hz, so we need longer settling time. Other applications may not face the same problem.

[image]

[image]

closed 2024-09-24 by adisuciu