analogdevicesinc/scopy #810: Scopy:: Can't change languages macOS

vincenttl, 2020-06-28:
**Environment:**
 - OS: macOS Mojave 10.14.6
 - https://github.com/analogdevicesinc/scopy/releases/tag/v1.1.2

Cannot change language in preference.

**To Reproduce**
Steps to reproduce the behaviour:
Last part of this page
https://wiki.analog.com/university/tools/m2k/scopy/translate_scopy

DanielGuramulta, 2020-06-29:
Hi

An issue with the translation feature might be that in v1.1.2 just some gui controls where added. Could you try one of our dev builds: https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/33753783/job/0lmbgsc99ny6aivd/artifacts and see if this works for you? Our next release should have the translation feature fully working.

The new ts files should be added in `
scopy/resources/translations/
`

vincenttl, 2020-06-29:
It works in this dev build. Thank you

closed 2020-06-29 by vincenttl