analogdevicesinc/scopy #812: Scope : Measurements : phase

rgetz, 2020-06-29:
Phase Measurements on scope channels.

In most time domain scopes, there is the ability to measure "phase" difference between two signals (or a single channel, and a reference or math channel).

https://www.ednasia.com/measure-phase-difference-with-an-oscilloscope-2/

[image]

https://www.tek.com/blog/basic-time-and-amplitude-measurements-tbs2000-oscilloscope-part-3-3-xyzs-series

[image]

I think Scopy is missing that.

Some pro/con of different measurement techniques:

http://eeshop.unl.edu/pdf/OscilloscopeTutorial--PhaseMeasurement.pdf
http://www.autex.spb.su/download/wavelet/books/sensor/CH41.PDF

tfcollins, 2020-06-29:
Phase info is definitely useful. Do we need a library to do the math on this? I don't think there is a straightforward way in GR to do this without a few blocks.

-Travis

damercer, 2020-07-07:
Phase often falls out of the other time measurements, frequency / period and where the edges fall relative to the trigger point and can be calculated from what is already there. That is how I do it.

Doug

rgetz, 2020-07-07:
Yes - you can manually calc phase difference with measurements/cursors.

Φ = td/tp × 360 
Where: td is the delay between waveforms and tp is the period of the waveforms.

It's just easier to have the computer do it for you. (it's always harder to have the cursors track two different waveforms) - measurements already track period, and "peak" (and I think "zero crossing") - so i should be straightforward to get a phase difference measurement...

damercer, 2020-07-07:
Of course I was just responding to the comment about needing some special library function to calculate it.

Doug

[External]

adisuciu, 2020-07-16:
The current measurement processing engine does not support inputs from multiple channels. We have to add that in order to compute phase (since phase is computed from 2 different waveforms). This will be useful for other measurements as well.

-Adrian