analogdevicesinc/scopy #813: phasor measurements

rgetz, 2020-06-29:
There are many times where you are interested in the difference between phase magnitude between two signals, and there is specialized instruments for that, known as a Phasor Analyzer. This is typically used in teaching power systems and power electronics circuits.

https://www.labvolt.com/solutions/2_electronics/98-9063-00_data_acquisition_and_control_interface
[image]

https://www.youtube.com/watch?v=6lSrt7Kp0lY
[image]

It is similar to the view that we already have for the nyquist plot, so hopefully we can re-use the widget from that.
[image]

The difference is the magnitude is just linear volts, not dB, and you are interested in showing Channel 1 and Channel 2, and measuring the difference between them. Channel 1 is always 0 degrees (as the base reference). I don't know if this is a completly separate instrument (I don't think so), handling it in the time domain view (similar to the X/Y plot) would be best (I think).

damercer, 2020-07-07:
Right it is called a vector voltmeter as well. Power engineers use displays like this for relative phase between voltage and current to measure power factor in AC systems.

Doug

rgetz, 2020-07-07:
yes, many are used in RF measurements as well...
- 2-port transmission, magnitude and phase
- B/A and A/B, magnitude and phase

For example: it's an advanced option on the [KeySight 2 port analyzer](https://www.keysight.com/en/pd-2423461-pn-N9928A-308/vector-voltmeter?cc=US&lc=eng), 
[image]

There is some interesting history in an [Anritsu App Note](https://dl.cdn-anritsu.com/en-us/test-measurement/files/Application-Notes/Application-Note/11410-00531.pdf)

rgetz, 2020-07-14:
comment from faculty was would be nice to have magnitude in Volts RMS, Volts Peak, and dB (relative to reference).