analogdevicesinc/scopy #817: all: line thickness

rgetz, 2020-07-10:
In the scope - we can control line thickness - that needs to be added to the other instruments as well.

-Robin

adisuciu, 2020-07-16:
The new logic analyzer and pattern generator have line thickness

The only instruments that are lacking are signal generator, DMM and spectrum analyzer.

One note on the spectrum analyzer is that the QWT plot is not really optimised for rendering thicker lines (I think that is why we didn't add it yet). There may be a big performance hit when enabling thickness. The oscilloscope is sluggish when running with maximum line thickness.

We're going to implement this and somewhere down the line we'll look into plot optimisations - we already tot some for the logic analyzer

-Adrian

mthoren-adi, 2020-08-11:
@teoperisanu , this should also be added to the X-Y display, we're using it in a curve tracer experiment and the default line thickness is pretty thin. The dots option would work, but they're too big (resulting line is thicker than the channel traces when set to 5). Maybe a single thickness drop-down that applied to both dots and lines, whichever is selected?

The FFT display in the Oscilloscope is probably okay as-is, if a user wanted to look at frequency domain in detail, they'd move over to the spectrum analyzer anyway.

adisuciu, 2020-08-14:
duplicate of https://github.com/analogdevicesinc/scopy/issues/817
closing

closed 2020-08-14 by adisuciu