analogdevicesinc/scopy #818: cannot build successful

Hansen0314, 2020-07-13:
Hi,
we got some trouble when building the project.
there is some error as below when we use  https://wiki.analog.com/university/tools/m2k/scopy/build-windows to building Scopy.
[image]
 and I have been finished step before  'cmake -GNinja $SCOPY_CMAKE_OPTS $CMAKE_OPTS ../'
  this is content of SCOPY_CMAKE_OPTS  and CMAKE_OPTS 
[image]
and the branch of scopy source code  is master
[image]
[image]
[image]

adisuciu, 2020-07-13:
It looks like log4cpp is not installed in your system. Can you confirm that you installed scopy-mingw-deps in the MSYS system succesfully ? (you ran  the following script without errors) 

```
source ../CI/appveyor/set_build_env_msys.sh
../install_msys_deps.sh
```

if yes, can you find the following file <your_msys_install_folder>/mingw64/bin/liblog4cpp.dll  ?

-Adrian

Hansen0314, 2020-07-21:
Yes, I have been installed scope-mingw-deps in the MSYS system successfully .and I cannot find the liblog4cpp.dll at /mingw64/bin/. and I got another error when I use the ubuntu to install the env. can you give me a correct building wiki?

Hansen0314, 2020-07-21:
and the ubuntu version is 18.04.
[image]

Hansen0314, 2020-07-22:
Hi @adisuciu  
do you have some idea about this question?

adisuciu, 2020-07-22:
So you want to install mingw or ubuntu ? 
For mingw please provide an output of install_msys_deps after running set_build_env.
For ubuntu i'll update the build guide today.

-Adrian

adisuciu, 2020-07-29:
I assume this was fixed since you sent out a PR

closed 2020-07-29 by adisuciu