analogdevicesinc/scopy #819: Signal Generator: Stairstep Waveform Type

amiclaus, 2020-07-13:
It would be nice to have the Stairstep implemented as a Waveform type.

The Scopy menu for this waveform type should be similar to the other basic waveform types + "number of steps" as an extra configurable parameter.

adisuciu, 2020-07-14:
We can create a stairstep function in the waveform type with the following parameters  (bold is my suggestion)
**amplitude p-p (of the whole stairstep)**   / amplitude of a single step
**frequency (of the waveform)** / sample rate (frequency of each step) / period (duration of each step)
**offset ( reference of first step)**
**phase (in steps)** 
**rising number of steps** 
**falling number of steps**

This might be a little more advanced (or maybe a separate feature), but another option would be to create a digitizing algorithm for all of the waveforms / buffers/ math . You can create stairstep by selecting rising sawtooth / falling sawtooth /triangle, and selecting min(offset) / max(amplitude) rails and number of steps inbetween.

You could use this to digitize stairsteps, sines or even WAV/CSV files, but it wouldn't be as straightforward
Thoughts ?

-Adrian

damercer, 2020-07-15:
I think what you suggest is adequate for the purpose at hand, i.e. the base/gate waveform of a curve tracer.
But, to that end, it might be easier to understand if the stair step was defined by, rather than an amplitude / offset,  a starting voltage level, an ending voltage level and the number of steps? If start voltage is less than end voltage stairs go up, if start voltage is greater than end voltage stairs go down ( for P type devices ). The frequency / phase can be as usual to make aligning stair treads with collector / drain triangle waveform simple. (Freq of triangle would be No. of steps X frequency of stair waveform)

Doug

rgetz, 2020-07-20:
Some ways others have done it.

[keysight](http://rfmw.em.keysight.com/bihelpfiles/n6705/webhelp/Content/__D_Using%20the%20Source%20Functions/30%20-%20Generating%20Arbitrary%20Waveforms.htm)

[image]

setting | Meaning
-- | --
Start Setting (I0, V0, P0, or R0) | The setting before the staircase.
End Setting (I1, V1, P1, or R1) | The setting after the final step. The difference between the start and end setting is divided equally between steps.
Delay (T0) | The delay after the trigger is received but before the staircase starts.
Step Time (T1) | The time to complete all staircase steps.
End Time (T2) | The time the output remains at the end setting after the staircase completes.
number of Steps | The total number of staircase steps.

adisuciu, 2020-07-29:
https://github.com/analogdevicesinc/scopy/pull/833 implemented this

tagoylo, 2020-10-21:
Stairstep is implemented in v1.2. Closing this item.

closed 2020-10-21 by tagoylo