analogdevicesinc/scopy #820: Scopy with ADALM2000 Voltmeter incorrect readings

markallie, 2020-07-14:
**Environment:**
 -Win 10 Pro
 - V 0.25
 - IIO versioon 0.18

**Describe the bug**
I have been testing Scopy with an M2K for use in simple experiments. I had a 10Vpp 1KHz Sin wave as well as a 5Vpp 1KHz sine wave sometimes with a DC offset. It seems the Scopy Voltmeter app cannot measure the DC voltage of a signal if there is significant AC and DC voltage there. Has anyone else noticed this or perhaps I am doing something wrong. I did test DC only and then the Voltmeter app gives the correct DC values. However when there is AC it seems to be totally random. I had a 1VDC offset with an AC component on the signal generator and I got voltmeter reading anywhere from - 5.XXX to + 5.XXXX V I got the different reading by stopping and running the app. I tried using the oscilloscope at the same time to see if the acquisition of the signals made a difference. It didn't seem to matter.

Peak hold on or off doesn't seem to change anything with the incorrect DC value.

The AC measurement values seem OK.

Mark Allie

**To Reproduce**
Steps to reproduce the behavior:
Run Scopy with an ADALM2000 connected to the PC.
Generate a 6Vpp 1KHz sine wave with 2VDC offset with the signal generator.
Connect the CH1 and or CH2 + and - inputs to the signal generator output. CHX+ to the output and CH- to GND
Start the voltmeter app. 
Click to display a DC (Direct Current).
The value given in the app is incorrect and seems random.
Go to AC (either one) and back to DC,
The value will change.
Change the DC value in the waveform.
The value will still be incorrect and apparently random.
Press stop and run'
The values randomly change.
The CH1 and CH2 values match within 10% if the same signal is applied to both inputs. Both wrong.

DC measure correctly when there is only a DC voltage present at the input.

AC seems to measure correctly no matter what the DC component is.

**Expected behavior**
The DC value from the voltmeter  would equal the DC value in the Signal Generator output.
[Scopy.ini.zip](https://github.com/analogdevicesinc/scopy/files/4919306/Scopy.ini.zip)

adisuciu, 2020-07-29:
https://github.com/analogdevicesinc/scopy/pull/825 should fix it

tagoylo, 2020-10-21:
This bug is fixed on v1.2. Closing this item.

closed 2020-10-21 by tagoylo