analogdevicesinc/scopy #821: Signal Generator: Math/equation interpretation

mthoren-adi, 2020-07-14:
**Environment:**
 - OS: all (assumed)
 - FW v0.25
 - SW v1.1.2 - 3d4c582

**Describe the bug**
The time variable "t" is actually interpreted as "2*pi*f*t", where f is the frequency from the dropdown menu.
Also, while the entered frequency shouldn't matter, setting too high or too low results in unpredictable behavior (see below)

**To Reproduce**
For the desired waveform:
(1+1*cos(1000*2*pi*t))*sin(9500000*2*pi*t)
you have to enter
(1+1*cos(1*t))*sin(9500*t), with a frequency setting of 1kHz.
And entering:
(1+1*cos(1000*t))*sin(9500000*t) with a frequency setting of 1Hz produces a corrupt output. Actually, it looks like this might be indicated by a red line under the equation, but this is subtle and isn't written down anywhere:
[image]

**Expected behavior**
Interpret time as time (requiring the user to enter 2*pi if that's what they want), and calculate an optimum sample rate (and buffer length) based on the entered equation. Ideally, change the "frequency" drop-down to display the resulting sample rate(not settable by the user).
It would also be good to have the ability to save the resulting data, tell what t he buffer is, etc. (nice to have, not essential.)
Also nice to have - a pi key on the keyboard. But it looks like the string "pi" is interpreted as pi.

mthoren-adi, 2020-07-24:
@adisuciu , So far things are looking great, I've thrown a bunch of strange functions at it and they are all working as expected. One thing I noticed is that when you set the number of displayed periods higher than 1 (in the main preferences screen), it looks like the preview screen just keeps time progressing across the horizontal axis, rather than displaying that number of data records concatenated together, which is what we want for easy visualization of what happens as the buffer rolls over and repeats. But the actual waveform produced is correct. In this screenshot, I want a half-wave rectified sinewave, so I set sinewave frequency to 500Hz, and a data record length of 1ms, and display 4 periods. Signal verified on the scope is correct, but the preview is wrong.

[image]

adisuciu, 2020-07-27:
Fixed with https://github.com/analogdevicesinc/scopy/commit/6d61d4c872ff34665bce5f7c25c7b6046d217186

[image]

adisuciu, 2020-07-29:
https://github.com/analogdevicesinc/scopy/pull/833 implemented this

HRosete, 2020-10-21:
This is fixed on Scopy v1.2 (verified on 33ca72f9). Closing this item.

closed 2020-10-21 by HRosete