analogdevicesinc/scopy #822: Scopy with the ADALM2000 (M2K) Plot labels (Educational Need)

markallie, 2020-07-14:
One cannot place a short label on the Oscilloscope plot screen after the image is there.

I would like to be able to place at least one word on the screen. Anywhere is desired but one place is acceptable.

I use this feature with other scope apps so the students can put their name on the captured scope image to prove they have done the capture themselves.

Other uses would be to label traces with useful names for documentation purposes.

rgetz, 2020-07-16:
@markallie : It shouldn't be a hard thing to do - but I think your students are sophisticated enough to use paint to replace a some else's in a png - aren't they? (that's why we haven't done it in the past).

Is the device serial number a better thing to use?

We could encode the device serial number as a QR Code in the background of the image. Similar to:

https://www.spiedigitallibrary.org/ContentImages/Proceedings/10443/1044317/FigureImages/00402_psisdg10443_1044317_page_3_1.jpg 

which is described in this paper:
https://www.spiedigitallibrary.org/conference-proceedings-of-spie/10443/1044317/Interactive-QR-code-beautification-with-full-background-image-embedding/10.1117/12.2280282.full?SSO=1

?

adisuciu, 2020-07-16:
I misread this as being able to set names for channels in the oscilloscope - something like this. Which I think is a good idea.
[image]

When exporting CSV files, the device serial number is already there. You can pair the CSV file with the exported plot - The waveforms should be pretty much the same.

-Adrian

markallie, 2020-07-16:
There is very little a student could not circumvent. I make it a name on the plot because it is usually easy to do and it is also usually easier to do the actual capture with your name on it as opposed to using a graphical editor to modify an image. A serial number as a background or foreground item would also work for my needs. I will look at those articles referenced within.

Thanks

Mark Allie

markallie, 2020-07-16:
Also I don't use this for testing. It is used for weekly assignments. If students use a graphical editor instead of doing the real iscope capture they just hurt themselves when it comes to the actual bench exam.

rgetz, 2020-07-21:
@adisuciu : How difficult would be to to be able to add something the allows (a) arbitrary text and (b) arbitrary text + arrows?

Similar to:
https://www.qcustomplot.com/index.php/tutorials/items

@markallie : Would that meet what you need to do?

Thanks
-Robin

BTW:
https://github.com/danielsanfr/qt-qrcode
https://github.com/nayuki/QR-Code-generator/tree/master/cpp
https://github.com/fukuchi/libqrencode

markallie, 2020-07-21:
The concept portrayed in the tutorials/items would achieve what I am after accept I need this in the app Scopy for the Analog Devices ADALM2000. This looks like it is for something e4lse. Is that true or am I misinterpreting what I am seeing?

 

Mark ALlie

 

https://www.qcustomplot.com/index.php/tutorials/items

https://github.com/danielsanfr/qt-qrcode
https://github.com/nayuki/QR-Code-generator/tree/master/cpp
https://github.com/fukuchi/libqrencode

rgetz, 2020-07-22:
@markallie : yes - this is just a Qt tutorial - since scopy is built on Qt, it's easier to point out existing open source implementations to see if that would work.

markallie, 2020-07-22:
Robin, I understand the open source concept and it is a good one. I on the other hand probably wont learn to create code for the Scopy app. Is the code you demonstrated what someone is going to do or is something I should do.

 

Mark Allie

markallie, 2020-07-22:
> @markallie <https://github.com/markallie> : We were looking for how others did things:
>
> *	Custom Waveform Labels <http://rfmw.em.keysight.com/DigitalPhotonics/flexdca/UG/Content/Users/Display/waveform_labels.htm>  on Keysight.
> *	Single line of Text <https://www.picotech.com/download/manuals/picoscope6-oscilloscope-software-users-guide.pdf#page=53>  on PicoScope
>
> We were just trying to understand how to expose things to end users?
>
> Thanks

Look at Waveforms to for label options.

 

Mark Allie

adisuciu, 2020-08-12:
Does this look alright ?
[image]

-Adrian

rgetz, 2020-08-12:
Thats "notes" at the bottom? Not "labels" in the actual plot?

adisuciu, 2020-08-12:
Yes, this was an easy implementation. Doesn't require any fancy plot operations. If this is enough, we can have this on all instruments by next week. 
I'll also sketchup an implementation with actual plot labels some time this week.

markallie, 2020-08-12:
I assume you mean the section marked insert notes here. Yes that would meet my needs perfectly. I assume it shows up if the app is separate from the Scopy window.

 

Mark Allie

 

 <https://user-images.githubusercontent.com/8676912/90035880-0515ad80-dccb-11ea-88c4-a6d94fe37912.png>

markallie, 2020-08-12:
I am sure this will meet all my needs and hopefully the needs of others as well.

 

Mark Allie

adisuciu, 2020-08-14:
I implemented the notes for all instruments. 
They will not be part of the printed plot however if you decide to use the built-in print-to-image function. 

[image]
will give out this image when using the print function
[image]

-Adrian

markallie, 2020-08-14:
My intended use will have the students capture the whole scope app and paste it into the document they are going to habd in. So it seems this will work fine for that. Say can I get a copy of this software version to play with? Thanks

 

Mark Allie

 

 <https://user-images.githubusercontent.com/8676912/90260904-e2170500-de54-11ea-81d5-2db7aa7c2545.png> 
 <https://user-images.githubusercontent.com/8676912/90261069-1db1cf00-de55-11ea-903a-fd8c400a8600.png>

adisuciu, 2020-08-17:
Yes, you can download a 64bit installer here: https://ci.appveyor.com/api/buildjobs/k96gogac71dqxvuk/artifacts/scopy-64-setup.exe

Make sure you delete Scopy.ini file - https://wiki.analog.com/university/tools/m2k/scopy-troubleshooting#ini_files

And restart your computer after installation. It's better(safer) to uninstall your current Scopy version before installing this.

-Adrian

markallie, 2020-08-17:
Thanks. I’ll uninstall for sure.

 

Mark

markallie, 2020-08-17:
I installed the new version of Scopy and I tried the print function. It does not give me what you included in your email. It must be some other thing to do that I don’t know about yet. How did iyou get the image in this email?

 

Mark Allie

 

 <https://user-images.githubusercontent.com/8676912/90260904-e2170500-de54-11ea-81d5-2db7aa7c2545.png> 
 <https://user-images.githubusercontent.com/8676912/90261069-1db1cf00-de55-11ea-903a-fd8c400a8600.png>

adisuciu, 2020-08-18:
First image is a regular screenshot using the snipping tool (for example) .. The second image is using the print button in the top left of the instrument.

-Adrian