analogdevicesinc/scopy #823: Connecting to virtual devices

balping, 2020-07-14:
Is it possible to use Scopy to connect to a virtual device, ie a software emulated oscilloscope that is supplying generated real-time raw acquisitions? I'm willing to modify the source if this is possible in theory. I looked a bit into the source and I have the feeling that it should be possible to emulate an iio device and communicate with Scopy. Is this far fetched? Has anyone done something similar?

rgetz, 2020-07-16:
@balping - I on't think anyone has looked at this.

Is it better to use a virtual device? or an audio sink/source?

rgetz, 2020-07-27:
Eventually - we need to be able to support audio in/out (soundcard)

https://www.codeproject.com/Articles/421287/Cross-Platform-Microphone-Audio-Processing-Utility

adisuciu, 2021-05-05:
ADALM2000 emulator implemented .. soundcard support should be a task for iio-emu

closed 2021-05-05 by adisuciu